TheZeroBeast/TZB-MMU2S-Firmware #96: Spurious "filament change" pauses

aarchiba, 2019-01-20:
I'm using "MMU-MK3 2.1.8-RC1 TZB 7x7" (though the previous release did this too) and every five or ten layers the print stops and the printer beeps. The LCD message indicates it thinks there's been a filament runout. The filament is generally retracted, the selector is moved aside, and the filament is advanced - there is a characteristic ~7 cm sticking out of the MMU2 body, and no filament in the Bowden. I clip the excess filament or just shove it back in, hit the button on the MK3, and the print resumes fine. This occurs with several colours (white, blue, green) that worked reliably before (with this firmware from ~3 releases ago).

TheZeroBeast, 2019-01-20:
As long as you're slicing with Slic3r-PE Alpha3 as per documentation this would be due to a Jam being detected which is a new feature in 2.1.6-8, if you're getting false positives please look into indirect filament sensor mods.

aarchiba, 2019-01-21:
This does sound like jam detection but I'm a little confused by the reference to a Slic3r version - is there a recommended Slic3r version for use with this code? Is this documented anywhere? I don't see it in the README.md. I'm using 1.41.2, though I don't see how this would relate to spurious jams.

aarchiba, 2019-01-24:
I am still getting a lot of spurious jams - to the point I can't print - even though I wasn't getting any skipped layers before. Switching to the alpha version of slic3r hasn't made any difference. I do plan to switch to an indirect filament sensor mod, but I'm not sure it will fix the problem if the jams are really spurious.

closed 2019-08-20 by dvisser