TheZeroBeast/TZB-MMU2S-Firmware #97: MMU2 debugging following filament runout problematic

aarchiba, 2019-01-20:
Because of #96 I am having numerous stops for filament change. If all goes well the print resumes, but if something goes wrong on load - for example if the "new" filament sticks too far out so the selector bends it over and it doesn't feed into the Bowden - I don't get an MMU2 debugging menu, all I get is the "Changed correctly?" menu, with the selector locked. As far as I can tell the only way to get out of this state is to reboot the MK3 and abandon the print. It would make more sense to enter the standard "please help me load filament N" state here. (The "is the color clear?" thing doesn't make sense either.) 

MMU-MK3 2.1.8-RC1 TZB 7x7

TheZeroBeast, 2019-01-20:
You will need to provide more information, I don't understand what issue you're having.

In any error state the LED will blink and if the FINDA detects filament buttons won't do anything until FINDA is manually cleared as per instructions in README.md

aarchiba, 2019-01-21:
> Aarchiba, the "is the colour clear" thing you are referring to is simply asking is the printer extruding the colour you intended it to. E.g If you intend to start printing in yellow, however you have previously printed with black there will be a bleed of black colour through the start of the extruded yellow. In that menu of "is the colour clear?" if you select 'No' your printer should extrude another lot of yellow filament. Continue to do so until the filament is completely yellow with no black bleed through it, when you are satisfied with the colour select 'Yes' and continue as you would normally.

Yes, of course. But when a filament runout has occurred this is not a helpful option.

TheZeroBeast, 2019-01-21:
Depending on how that was initiated it is definitely a helpful option as it could be a Jam that was detected that then needs to be purged. In any case it is Prusa stock code which has that message and you can start an issue with them if you don't think it's relevant.

Can you please provide further details, logs and gcode files for the specific issues you're having so I can attempt to assist you with them?

TheZeroBeast, 2019-01-22:
Closing due to lack of communication.

closed 2019-08-20 by dvisser