TheZeroBeast/TZB-MMU2S-Firmware #99: Clarification on this firmware working on MMU2 / MK2S or MK2.5

eschlenz, 2019-01-23:
My apologies for putting this question in the Issues section. I wasn't sure how else to ask this question.

If I were to flash just the MMU2 with this firmware, will the stock firmware for the MK2S or MK2.5 be compatible? Or does the MMU2 firmware you developed only work if you have an MK3, and use your MK3 firmware as well?

TheZeroBeast, 2019-01-23:
the entire communications protocol has been replaced so no talking between and I haven't done any coding for 12v supply so MMU2 software wouldn't work as intended.
If there's a avr programmer that has a 2.5MMU2 they could do the required code to make it happen but I don't have the time to at present.

An option is the run my mk3 code compiled for MK2.5 and then have a 24v 5a supply powering the mmu2 then it will work as if on a MK3.

eschlenz, 2019-01-23:
OK. Thanks for the reply @TheZeroBeast. I'm going to open a PR that updates your README.md, to indicate that this is not compatible with the MK2S/MK2.5 at this time. That way others in my position might be able to find that information a little easier.

closed 2019-08-20 by dvisser