TheZeroBeast/TZB-MMU2S-Firmware #103: "Remove old filament and press the knob", followed by idler hammering endlessly

jettoblack, 2019-01-26:
I'm using 2.1.8-302 (built myself) and 2386. Using slic3r 1.42.0-alpha3 to slice and Octoprint to print. I am using a BMG-like extruder, but not sure if that has anything to do with this. I was previously using 2.1.5 with no issues.

First 5-color test print went fine, no issues.

Second MMU print started ok, the purge line printed all 5 colors successfully. 

However at the first color change, the MK3 LCD displayed: "Remove old filament and press the knob to start loading new filament." (Never seen this when using the MMU before.) The MMU was not in an error state, it had a green solid light on the previous filament (F5), and no other LEDs were on or blinking. The F5 filament was sticking out the front of the MMU a few inches, while the selector was on F1 (the filament to change to). The filament for F1 was not loaded into the selector at all, it was still in the tube.

Not sure what to do as this situation isn't described in the README, I manually pulled back the part of F5 that was sticking out the MMU, and then pressed the knob on the MK3. Now, the MMU signaled an error condition loading F1. So I hit the middle button on the MMU to try to continue loading F1.

Next, the idler on the MMU started hammering back and forth very hard, in an endless loop. I took a video of this, it went on for more than a minute before I powered off the printer.

I'm not sure what happened but I'm guessing the initial load failure was caused by the idler getting out of sync and trying to push F5 instead of F1, but I'm not sure what caused the endless hammering of the idler. I opened up the idler and checked that nothing was jammed, it still rotates smoothly by hand, it doesn't seem to be hitting any resistance and nothing appears to be cracked or damaged in any way.

jettoblack, 2019-01-26:
Now when I turn the printer on, I get the endless idler homing loop at startup. Here’s a video:
https://youtu.be/7PSimWH1dos

jettoblack, 2019-01-26:
The idler homes correctly when the top is opened up all the way. I only tighten the tension screws until they are flush, no more. They are definitely not excessively tight, if anything they are looser than I'd normally tighten them for proper behavior in TZB 2.1.5, but in 2.1.8 the idler still gets stuck in a loop trying to home.

TheZeroBeast, 2019-01-27:
Hey @jettoblack, nice video, thank you 👍

The first post about the ejected filament is Jam detection which has now changed in the latest release so go check that out too.

Can you try again with the latest commits please, I was mid dev on some sg work on that commit.
If selector or idler stall it will now only do it 3 times before stopping for some human assistance.

TheZeroBeast, 2019-01-28:
Hey @jettoblack,

Can you please review the latest release as there are Jam Detection changes that will help.

jettoblack, 2019-02-05:
So far, so good on 2.1.8-RC2. I haven't been able to reproduce this kind of problem anymore.

closed 2019-08-20 by dvisser