TheZeroBeast/TZB-MMU2S-Firmware #106: new firmware, now printer and MMU don't communicate

YoungR0se, 2019-01-28:


TheZeroBeast, 2019-01-29:
Must be a bug, please provide log files and gcode for me to progress this investigation when possible.

YoungR0se, 2019-01-29:
Hi,
Thanks for taking the time to look into this :) I don't know how to provide gcode, so attached are the firmware files that I used.  The MMU is unaltered, the printer has Z axis increased to 350mm.  
In addition to yesterdays info, I reflashed the firmware today while I was preparing this for you.  When I preheated the printer on the screen between 'SD%' and 'time remaining', it says 'F83'.  When temp was attained, I tried to load Filament 1, which was successful.  Then I tried to load Filament 2, which was not.  I selected Filament 2 from the loading menu and nothing happened.  Upon returning to the home screen, the 'F83' has now become '83[right arrow]1, and the status bar says 'MMU Loading Ext: 1'.  

[30-01-19 Logfile MK3 B 2386.docx](https://github.com/TheZeroBeast/MM-control-01/files/2810018/30-01-19.Logfile.MK3.B.2386.docx)
[30-01-19 Log file MMU V 2.1.8 RC2 B 304.docx](https://github.com/TheZeroBeast/MM-control-01/files/2810019/30-01-19.Log.file.MMU.V.2.1.8.RC2.B.304.docx)

[Attempted flash of MMU V 2.1.8 RC2 B 304 firmware.zip](https://github.com/TheZeroBeast/MM-control-01/files/2810022/Attempted.flash.of.MMU.V.2.1.8.RC2.B.304.firmware.zip)
[Attempted flash of MK3 B 2386 firmware.zip](https://github.com/TheZeroBeast/MM-control-01/files/2810023/Attempted.flash.of.MK3.B.2386.firmware.zip)

YoungR0se, 2019-02-11:
Yo, it's been two weeks.  Anything changed?  Bug fixed?  I've kept an eye on the page but no new files have been added, the most recent was 16 days ago.

TheZeroBeast, 2019-02-11:
Hi @youngr0se,

I had a quick look at your provided files last night and couldn’t find any smoking gun.

There’s no obvious reason why communications wouldn’t be working, please check your hardware.

YoungR0se, 2019-02-12:
Ok, thanks for the info

AbeFM, 2019-02-21:
Not sure if you've figured it out, but have you checked that your sensor is seeing your filament reliably? Certainly that was an issue that had caused similar behavior in the past for me.

TheZeroBeast, 2019-08-20:
Stale. Closing

closed 2019-08-20 by dvisser