TheZeroBeast/TZB-MMU2S-Firmware #107: Octopi No connect Serial Port

gizzburn, 2019-01-30:
Robert I have been having problems when using Octoprint to connect to my printer, this only started with the 7x7 addition to the firmware I think.
I have done some testing today with it, first I did a clean install of Octoprint, no change it takes a long while to connect or just does not connect.

I have another printer with the same setup running on the same pi, so I used this with a new clean setup of octoprint  to connect it my Mk3 with the same results as my other pi.

I then did a a two pass flash with your latest firmware incase it was a bad flash, still the same result.

I then loaded your Mk3 5.0.4-TZB-BN2370 firmware and it connects every time now, I have seen a few posts on FB with guys having the problem and they also mention that they think its something with the 7x7 mesh code.

any ideas ?.

gizzburn, 2019-01-30:
Robert update just installed MMU: V#: 2.1.8 RC2 B#: 304
MK3: B#: 2386  and it connects now every time but after 20 to 30 seconds but does connect.

TheZeroBeast, 2019-02-02:
Hi @gizzburn,

I'm pretty sure this isn't anything to do with the changes I have made as I have tested both stock and this fw with the same outcome. If you do find a reason please open an issue with Prusa Firmware so it can be patched for all.

closed 2019-08-20 by dvisser