TheZeroBeast/TZB-MMU2S-Firmware #108: Help with Calibration

sergiofagundes, 2019-01-30:
I have followed the BOWDEN_LENGTH calibration setup, everything goes well till step vi.

With step vii: I press the middle button and the MMU2 unload the filament (and begin to flash a red led).

I press now the left button to save and exit and it doesn't save and exit, it just act like feeding the filament.

Also the Filament mm is get every calibration increasing more and more

sergiofagundes, 2019-01-31:
Ops! I have used a LONG bowden tube (about 700mm long). I have cutted it to a 400mm and now the calibration worked perfectly.

Now the question is another, is there any settings to allow for a longer bowden tube in the firmware ?

TheZeroBeast, 2019-02-02:
Ok, so I found what is wrong and I'll patch for RC3.

Simple put I'm not unloading the newly calibrated length instead using the old length which if distance increased will never unload enough and fail.

TheZeroBeast, 2019-03-24:
Hi @sergiofagundes, this bug is fixed in new release

closed 2019-08-20 by dvisser