TheZeroBeast/TZB-MMU2S-Firmware #110: Unload Failure

sergiofagundes, 2019-02-05:
I was using "First Layer Cal" and everything goes well, but when it finishes the MMU2 unload the filament and get a Unload failure message on LCD.

So I press the MMU2 middle button and after the right button to complete the unload, but it only unloads if I force pushing the filament.

When calibrating the MMU2 bowden length the load/unload performs perfectly with no failure, this issue only happen after the filament pass the PINDA and goes through the bondtech gears (witch I have calibrated earlier to the middle of the gears).

How do I fix this ?

cavotechnology, 2019-02-05:
> I was using "First Layer Cal" and everything goes well, but when it finishes the MMU2 unloads the filament and get a Unload failure message on LCD.
> 
> So I press the MMU2 middle button and after the right button to complete the unload, but it only unloads if I force pushing the filament.
> 
> When calibrating the MMU2 bowden length the load/unload performs perfectly with no failure, this issue only happen after the filament pass the PINDA and goes through the bondtech gears (witch I have calibrated earlier to the middle of the gears).
> 
> How do I fix this ?

Me too. I have a Bondtech BMG extruder, pancake engine LDO 0.9.

sergiofagundes, 2019-02-09:
I have got a sucessfull unload after I fine tune the MMU2 springs screws. But it only works when using just 1 filament (in any position 1-5). If I have more filament loaded the unload fails. Looks like the idler doesn't get well aligned when there is 2 or more filaments loaded.

Maybe I need to increase the TMC2130_SG_THR_IDL ?

sergiofagundes, 2019-02-09:
Again, after I fine tuning the bowden length and bondtech length (because I played with the MMU2 tensioner springs) it working better now with all filaments loaded.

So investigating more on this, I found the the tip of filament on unloading is the cause of the issue of Unload failures.

Now I need to get a way to get the filament tip out more uniform to fix this issue. Any tips?

TheZeroBeast, 2019-02-11:
May be a bug I have overlooked in the cal. Printing wouldn't be affected.

sergiofagundes, 2019-02-11:
Finally I got my first prints sucessfully. I have change the bondtech length  (not in the middle of the gears) but just a little above the middle gears.

AbeFM, 2019-02-21:
Sounds like this issue could close...

closed 2019-08-20 by dvisser