TheZeroBeast/TZB-MMU2S-Firmware #111: OctoPi Not Connecting

eboston, 2019-02-08:
I just installed the latest version of the code and can print find from SD card.  However, OctoPi won't connect.  I saw another closed issue where updating to the latest code fixed their problem.  However, I still cannot get connect using the same version they did.  Is there something that needs to be changed in OctoPrint to get it to work?

Connecting to: /dev/ttyACM0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x64eebbb0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: echo: 2.1.8-TZB77-2386
Recv: echo: Last Updated: Jan 27 2019 16:45:43 | Author: (none, default config)
Recv: Compiled: Jan 27 2019
Recv: echo: Free Memory: 1851  PlannerBufferBytes: 1392
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: CrashDetect DISABLED
Recv: PAT9125_RES_X=0
Recv: PAT9125_RES_Y=240
Recv: PAT9125_init:1
Recv: PAT9125_RES_X=0
Recv: PAT9125_RES_Y=240
Recv: PAT9125_init:1
Recv: FSensor ENABLED
Recv: echo:SD card ok
Recv: MMU => MK3 '218'
Recv: MMU => MK3 '304'
Recv: MMU version valid
Recv: MMU => MK3 'Active Extruder:2'
Recv: MMU - ENABLED
Recv: PAT9125_RES_X=0
Recv: PAT9125_RES_Y=240
Recv: PAT9125_init:1
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor

Allram, 2019-02-08:
Have the same problem on the 2.1.6 version. I just press the reset button on the printer enough times so that it finally connects. I can't seem to see in the log what Octoprint actually waits on, it has connection to the printer as the logs are showing that.

gizzburn, 2019-02-08:
There are a few issues with the same problem being posted on the FB groups, I have been having issues connecting as well, so I think there is something with the 7x7 patch thats affecting the connections.

TheZeroBeast, 2019-02-11:
Still no sign that this is anything on my code changes. If anyone has evidence and logs to say otherwise I will revisit. Closing for now.

closed 2019-08-20 by dvisser