TheZeroBeast/TZB-MMU2S-Firmware #112: Question about motor currents

sergiofagundes, 2019-02-09:
What the numbers settings in currents for pulley, selector and idler means ?
In the following codes for example:

// currents for pulley, selector and idler
#define CURRENT_HOLDING_STEALTH        { 1,  7, 16}
#define CURRENT_HOLDING_NORMAL         { 1, 10, 10}
#define CURRENT_HOLDING_NORMAL_LOADING { 1, 10, 40}
#define CURRENT_RUNNING_STEALTH        {30, 35, 35}
#define CURRENT_RUNNING_NORMAL         {30, 35, 40}
#define CURRENT_HOMING                 { 1, 35, 35}

TheZeroBeast, 2019-02-11:
the current amount used to drive the motors at different times.

sergiofagundes, 2019-02-11:
> the current amount used to drive the motors at different times.

Hi, how to set for example a pulley stepper for a 500mA ? How these numbers are related to the mA ?
Thank you

TheZeroBeast, 2019-08-20:
Stale. Closing

closed 2019-08-20 by dvisser