TheZeroBeast/TZB-MMU2S-Firmware #113: Help with Slic3r Toolchange with Single extruder MM printers

sergiofagundes, 2019-02-10:
I'm using a hotend (not a full metal hotend) but a heatbreak with a ptfe tube (4.2mm/1.9mm) all through it.

What's the best settings for the toolchange parameters and Ramming settings to get a better filament tip on unload?

adlamy, 2019-02-11:
I don't think this is the best place to ask this kind of question, see https://shop.prusa3d.com/forum/original-prusa-i3-mk2-5-mk3-multi-material-2-0-f52/?sid=b1d79aae6942f9392d2f8644f2b5b123

TheZeroBeast, 2019-02-11:
Please ask in the forum, there may be someone that could assist.
I personally don't think it's a good idea as the full metal heatbreak is used for cooling. PTFE will deform in a matter of changes with molten plastic on it.
This is what happens to the 1.85mm customer PTFE in the MMU2 hotend when unload happens premature to proper cooling.

closed 2019-08-20 by dvisser