TheZeroBeast/TZB-MMU2S-Firmware #115: Prusa Mk3s Updates

gizzburn, 2019-02-13:
Hi Robert does the updates from Prusa with the new sensor create any problems for us using your firmware or will you tweak the firmware once it becomes known about the changes, Just curious about thi new sensor they have come up with, I presume you also will get the update having a MMU2.

sergiofagundes, 2019-02-22:
I'm curious too about this new MK3S extruder body and sensor and if it will work with TZB firmware.

Can't wait to use the new extruder, looking on how it works it looks much better than the old one.

AbeFM, 2019-02-22:
I've been using the Skelestruder with indirect sensor with some luck. I'm
having more trouble tuning my tips than I did before, trying to prove that
it works before I get to the latest ZB firmware... I'm pretty close to
there and don't have a feeling for how stable the current stuff is.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

TheZeroBeast, 2019-03-24:
Hi @gizzburn,

If anyone upgrades to the 'S' sensor then this firmware currently will stop functioning.
I have been thinking on what direction I should take for the future and I most likely will use the latest PR firmwares as a base and pull in some desired features from mine.

closed 2019-08-20 by dvisser