TheZeroBeast/TZB-MMU2S-Firmware #116: Briefly Excited...Then No.

AbeFM, 2019-02-21:
So who else thought the announced Prusa filament sensor would be monitoring the idler door?

It sure seems using something akin to what we're using now for firmware and a sensor on the idler would allow you to shoot filament towards the preemptively running extruder gears, and when they kick open,  you know you have the tip, and how far in you're already gone.

I think it's been discussed already. You could feed the filament blindly from MMU some rough preset distance, or even use original/indirect optical sensor to slow filament to sane speed before finding gears. If needed.

Mjankor, 2019-02-28:
There's actually some interesting possibilities here. The sensor cable goes to a 5 pin plug at the einsy, which I'm defining as pin 1 on the left, and pin 5 on the right. 

The on/off sensor uses power lines (pins 4 and 5), and pin 1 for sensing (based on cable colours). 

The older movement activated sensor uses pins 4 and 5 for power, and pins 2 and 3 for comms.

This means that using my extruder body (which holds both an original sensor + bearing proxy) and the MMU sensor tower) would allow for both sensors to be used at the same time. You could run a 5 wire cable with pin 1, 4, 5 for the old sensor, and pins 2,3,4,5 for the new sensor, with a split for the power at the extruder.

Not saying this is a good idea, I'm not even sure it's necessary or worth the effort involved, but it's an interesting concept.

AbeFM, 2019-02-28:
When everything's tuned perfectly, the machine will run blind. But I think
that would be a valuable piece of info. At least there's one less variable
involved.

Similarly, I think of could watch the MMU's sensor to look for blobs. You
can see the ball fall at the string, then rise again for the blob. I'd
rather know about it and stop then push it into the tube to cause a skipped
layer later.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

closed 2019-08-20 by dvisser