TheZeroBeast/TZB-MMU2S-Firmware #117: Project Status?

AbeFM, 2019-03-20:
Howdy!
It's been a while since I've experimented with the latest build - I've been using: 
Printer: 5.0.4-2330
MMU2: 2.1.5-260

I'd been sticking with these as I'd seen issues with the more current stuff stopping in the middle of the print. I'd been waiting for the next "stable" release to use as a permanent solution.... and it seems like perhaps work has stopped and it makes sense to compare the best ZB version to the current PR version and make a decision.

Does anyone have any insights for me along these lines?

TheZeroBeast, 2019-03-24:
There is no reports my latest version is stopping midpoint so I don't understand why you insist on running an outdated version.

I am looking into merging the latest PR firmware with upgrades from my code as I see fit.

If you or anyone else has had experience with the 'S' upgrades and the latest PR firmwares please advise as I haven't personally used them.

gizzburn, 2019-03-24:
@ TheZeroBeast  
My Printer is working just fine at the moment, I have a Bondtech extruder still in the box and have not got my Mk3s upgrades from Prusa yet so not in a rush to change anything at the moment.
I like the way your thinking of going with the firmware idea so can wait till you get it sorted before I change anything.
I have no problems with your latest version its been very stable for me, not had any prints stop mid print, If I have had a problem its been my fault.

AbeFM, 2019-03-25:
Good! I'm happy to be wrong about things. I was having some sensor issues around the time your new firmware came out, I haven't tried updating it. The Skelestruder introduced more modifications (some of those have been updated out) so really, when my replacement parts show up I'm anxious to try this again!

Again, thanks to everyone for letting me know I'm off base on this the-firmware's-broken thing! :-)

If you're going to be doing another round of incorporating the new PR changes (7x7 MBL, power supply non-clicking), I'd be happy to check them out in a few days (or later!) once things are back together.

closed 2019-08-20 by dvisser