TheZeroBeast/TZB-MMU2S-Firmware #118: 3.7 FW porting

AbeFM, 2019-04-05:
I finally got my hardware all fixed, and put on the OEM firmware (3.7.0 release, the latest), and OMG I forgot how slow and frustrating it is.

While I have to say it's pretty awesome that it still prints once hairs plug the sensors up, since it doesn't use them..... It's not as good as my ZBFW where I could eliminate grinding consistently.

Is a merge/port forthcoming? I would welcome one. I'd like to have an idea if it is on the way before I waste the time it takes to make the Prusa approach reliable.

TheZeroBeast, 2019-04-10:
Hi AbeFM,

They are too far apart for any easy merge which is why it's still coming.

I too tried the 3.7.0 release and the MMU2 code is even slower than when MMU2 was first released, it wasn't a nice experience.

Hopefully Prusa will ship my MMU2S upgrade this year and I can work on having something setup for that config. Even though it doesn't look reliable imho.

AbeFM, 2019-04-10:
Thanks for getting back - I'll continue to hold out hope. :-) 

I'm still not sure if they are measuring the idler door position or not. One 'press release' said it was being used, but that's all I've found.

AbeFM, 2019-04-10:
I saw the latest release, but have to modify printer code for my extruder. Do you have the mk3 source to compliment the mm-control-01-2.1.8 just released?

TheZeroBeast, 2019-04-10:
Same code as RC2 man 👍👌

AbeFM, 2019-04-10:
You mean https://github.com/TheZeroBeast/Prusa-Firmware/archive/5.0.3.zip which came from the (current) top of https://github.com/TheZeroBeast/Prusa-Firmware/releases ?

Knowing what goes with what is always a challenge.

TheZeroBeast, 2019-04-11:
The MK3 FW hasn’t changed on the 2.1.8 release, you can keep your current modded install and just update your mmu2 code.

AbeFM, 2019-04-12:
The last thing I have has this:
#define FW_VERSION "5.0.4-AbeSkele"
#define FW_COMMIT_NR   2330

AbeFM, 2019-04-12:
Latest stable versions & link to RELEASES for Precompiled HEX files.
MMU: V#: 2.1.8 RC2	B#: 304
MK3: B#: 2386

https://github.com/TheZeroBeast/MM-control-01/releases

Right there at the top of the ReadMe.

AbeFM, 2019-05-21:
So yeah, I got the new hardware from Prusa installed, and running their
latest firmware and software.

It's as shitty as it used to be - slow, unreliable, etc.

But it's all made up for by reading the state of the idler - admittedly,
I've only done two prints so far, but it works! I really miss all the fast
moves, responsiveness, etc from the earlier ZB firmware. And I'd love to
get rid of the print speed going to 30% every filament change.... and I
REALLY don't like how once in a while it ejects filament the wrong way,
grinding and making a blob on the purge tower INSTEAD of unloading.....

While loading, it kinda feeds the filament a couple inches, then pulls it
back, and the extruder gears are running backwards at high speed but
occasionally forwards at low speed, so sometimes it take a few tries (pulls
filament back to MMU and starts over).....

I don't know if you've taken much of a look at it, but it most certainly
could use your secret sauce now that there's hardware which should give an
unambiguous signal for when the filament arrived.

TheZeroBeast, 2019-08-20:
Working on adding new features and supporting MK3S-MMU2S machine.

closed 2019-08-20 by dvisser

AbeFM, 2019-08-20:
That's excellent to see this may get some clean up. I think it's less a matter of adding new "features" and more an issue of fixing glaring bugs. :-)