TheZeroBeast/TZB-MMU2S-Firmware #119: Question about MK3S IR sensor

YoungR0se, 2019-04-13:
Hi,
Are you going to make a subsidiary ZeroBeast firmware that supports Prusa's latest IR filament sensor?  
I ask only because I'm curious whether their new sensor is actually an improvement, but I can't compile Prusa's MK3S firmware.  If you have support for the IR sensor in the pipeline, I'll wait for it, otherwise I'm going to shelve Prusa's new design and go back the using MK3 (no S) ZeroBeast firmware, which I think is way better anyway :)

TheZeroBeast, 2019-04-23:
Hi @YoungR0se
In standard PRUSA fashion when I ordered my MMU2s upgrade in the first few hours of being live I'm still waiting for shipping. Last week I was advised that it'll be mid-may before it'll ship.

You ask it the sensor is better, in my opinion it is a massive step backward as we loose any way to sense for jams conditions. I have heard from a couple people that the IR sensor isn't foolproof, ie. you need to tweak it a fair bit until it is stable.

I am working on pulling what I can from my code into the latest PRUSA code, doing this I will leave behind a bunch of improvements that will no longer have a place with the new dumb sensor.

For the time being I am using my last release without issue

AbeFM, 2019-04-25:
I'd really like to see BOTH sensors used. There are ports, wires,
everything needed to do it. Retain the sensor for the filaments it sees,
turn it off (or use indirect) for the ones it doesn't, but taking advantage
of knowing when the filament reached the extruder seems like a win? If
their set up is unreliable, well, can you be surprised? It SHOULD be
possible to make it well.

Also, try hassling the PR guys via chat, you might get them to send it out
sooner. One thing is obvious - for all the belittling and
closed-mouthedness they do, they clearly watch the forums and use others'
improvements as their own - though rarely do they do that well. You may
well get your parts if you impress upon them you'll be doing development
with them?

Mine are shipping today.

lmyvgta, 2019-08-13:
Hi! Will you make an update for mk3s/mmu2s?

TheZeroBeast, 2019-08-16:
Hi @lmyvgta, I'm still finding the time to progress but still plan to bring new features to the code from PRUSA Repo.

lmyvgta, 2019-08-17:
Thanks I'll be waiting!

TheZeroBeast, 2019-08-20:
Working on adding new features and supporting MK3S-MMU2S machine.

closed 2019-08-20 by dvisser