TheZeroBeast/TZB-MMU2S-Firmware #120: How are people liking Prusa FW?

AbeFM, 2019-07-17:
I am sitting here, by my printer, because for the 5th time in a row, it's happily printing away with both of the filament sensors reading "empty". It didn't bother to load the filament, because it didn't bother to check if the extruder IR sensor was off before feeding filament.
It didn't care that before the filament sensor in the MMU triggered, it had already magically "arrived" in the extruder.

My motors still get hot, make noise.

The other day I came in, found the printer had failed it first material change, but somehow kept printing empty. Only when Z steps happened, they were ignored, so the printer spent an hour ironing the first color of the first layer.

Print times, etc, are displayed wrong. In short, there's a zillion errors, some of which you fixed before... And Prusa has their fingers shoved as far in their ears as always.

So - mostly I just want to make a list of issues here with sensible people who will at least verify/challenge what I'm seeing. I can't trust Prusa a half an inch.... But the 7x7 MBL and moved sensor are indeed capable of doing good prints...

Anyway, curious what others are seeing/thinking.

jettoblack, 2019-07-17:
I just recently switched from TZB to a mostly stock MK3S/MMU2S setup, and from a BMG-like extruder to a hybrid of the MK3S R4 extruder with Bear X-axis ends and Bunny Science gearbox for MK3S R4 extruder. The reason for the hardware change is that taking apart the BMG extruder to clear a jam required a near complete hotend disassembly, while the MK3S extruder is relatively easy to take the motor and front off. The reason for the software change was that I was tired of being behind on Prusa FW improvements that are not in TZB, such as the heatbed PWM improvements and the 7x7 leveling with magnet compensation, as well as the fact that stock MMU2(S) settings in PrusaSlicer don't work with TZB FW. 

I completed several test prints on the MMU2S including a 40 hour 4 color print with 1 load failure which was detected and recovered without any print artifacts, and 1 load failure caused by my auto rewind spool holder's spring breaking and tangling up the filament after it had already been loaded to the nozzle, which did result in a partial skipped layer but this is not the fault of the MMU2S, and TZB FW would probably not have helped here either.

Cons: the MMU2S with Prusa FW feels very slow compared to TZB, especially when homing and movement of the idler/selector feel way slower. Filament changes are just slightly slower. The TZB firmware had better troubleshooting and error recovery options using the buttons on the MMU2, the Prusa FW is kind of finicky and inconsistent when it comes to using the buttons to recover from errors. I wish the Prusa FW would power down the motors when the MMU is idle like TZB does. 

Overall I'd say I'm currently happier with the MMU2S than I was with the MMU2 with TZB. I think the best way forward would be to try to port some of the TZB improvements over to the MMU2S firmware but try to keep everything closer to stock, that way there is at least some chance that Prusa might merge the changes.

TheZeroBeast, 2019-07-18:
Hey Guys,

I've been using the MMU2S FW for a month now to evaluate and for the most part it works but they are a long way from where I was with this FW in regards to MMU2 use.

I have less than no time and it keeps getting re-allocated to other projects and until PRUSA decide to accept and recognise community efforts such as this we'll always be left behind.

My idea if time or assistance was found would be to keep my MMU2 FW and add it into the MK3S codebase. So much easy than done as there have been massive changes.

And if done and not merged would be another waste of 100s of hours.

AbeFM, 2019-07-18:
I completely understand and agree. While the community at large will have a huge condemning hissy fit if you say it outload, Prusa has not demonstrated an openness to contributions - so much so that if someone shows them a fix, they may steal it, but in not working with those who bring it up, they often miss the mark and release something poorly thought out and only tested by one guy. One guy testing something ten times is not the same as ten guys doing it once.

Anyway, I sort of figured that's where you're at - it is the reasonable position. Nonetheless, with so much room for improvement, I'm curious what people were noticing.

I'm not quite sure what things you've done would still apply, lots of details in serial communications?

Regardless, thanks for your earlier work and continued attention - just knowing that the machine doesn't HAVE to be slow/buggy/hot/noisy/unreliable is an improvement.

TheZeroBeast, 2019-08-20:
Stale. Closing

closed 2019-08-20 by dvisser