TheZeroBeast/TZB-MMU2S-Firmware #121: Are you back Mr TheZeroBeast??

dvisser, 2019-08-20:
Does this mean you are back and are adding support for all the latest hardware from Prusa, plus their current features and then more of your own?

This is epic if it is true! 

Welcome back TheZeroBeast!

TheZeroBeast, 2019-08-20:
It sure does @dvisser,

I'm sick of the painful PRUSA FW and have a little time to work some magic.

**EPIC!!!!**

dvisser, 2019-08-20:
I am sure you have some "surprises" up your sleeve! Welcome back!

TheZeroBeast, 2019-08-20:
Thank you @dvisser, many sleeves indeed....

AbeFM, 2019-08-20:
I, too, am super excited about this. While my new Zen approach of "just
live with what Prusa gives you" has been working, it eats at my soul. :-)

Mjankor, 2019-08-23:
What changes are you considering? Anything special with regard to sensors?

AbeFM, 2019-08-23:
Certainly things like "if the tube was 300 mm last time, and it doesn't load, we should feed 4 meters this time" are annoying. :-)

Similarly, if "waiting" for the door to trigger happens in < 1 mm from the MMU's sensor triggering, it's an error, not a sign of a proper load.

In general, any time the tube length comes back more than 50% different than it was last time, probably it's a sensor error and not the printer changing shape mid print.

Softer homing, reliable moves, etc. They all seem like ZB magic, let alone fixing serial coms and other under-the-hood stuff.

TheZeroBeast, 2019-08-24:
@Mjankor,
First I will be making my FW compatible with MK3S/MMU2S hardware then improving where I see fit. Previous benefits as mentioned by @AbeFM and more will be available when I release.

Mjankor, 2019-08-26:
Thoughts on jam detection?

TheZeroBeast, 2019-08-26:
My last FW Release had very stable jam detection although I haven’t started adding back the laser sensor it’s a good option to have best of both worlds. See how dev goes with that in the future.
@mjankor, are you having a lot of jamming with the stock MK3S FW?

Mjankor, 2019-08-26:
I'm still running the old 2.1.8 firmware, with a Mk3/Mk3s hybrid extruder (where I can choose to mount either the new IR sensor in the tower, or the old Mk3 sensor in the top-plate). I'm currently using the Mk3 sensor in indirect sensing mode (with bearing).

kajsilander78, 2019-08-31:
Is the MK3S/MMU2S version functional yet? Could not find any binaries... Just got the printer and have attached the MMU2S but cant stand the steps skipping during homing with original fw.....

TheZeroBeast, 2019-08-31:
Hi @kajsilander78,

Not as yet but I'm making progress, I'll have a release when it's ready for more robust testing 👍
I also can't wait to have my smooth roller back.

Mjankor, 2019-09-04:
How feasible is it to add in jam detection using a MK3 sensor mounted in between the two IR sensors (door and Finda)? 

Hardware-wise it should not be that challenging. I've already got an extruder than can mount both sensors, and they use different sensor/control pins at the Einsy.

I'm not expecting this to be a viable approach, but I'm still interested in how feasible it would be.

AbeFM, 2019-09-04:
My feeling has been replace the upper sensor with the mouse sensor, when
the ball goes up you can see it....

When the machine boots I have to press three or four buttons on the MMU to
get it to load filament. Since the head moves around, it may load the wrong
channel. If that channel is empty you can come back fifteen minutes later
to find it suck, still feeding the empty channel and ignoring user inputs.

Maybe after 30 seconds one can assume it isn't going to work?

It's designed by monkies.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

TheZeroBeast, 2019-09-04:
@Mjankor,

It’lol come down to how well it can be incorporated in code but hardware wise it’s just one extra cable to the extruder and a slight redesign of the extruder to house both sensors. I’d probably still go direct to limit part count and resistance on the filament.

@AbeFM, sounds like you’re enjoying the stock firmware ;-)

AbeFM, 2019-09-04:
Really I just need to improve the tall sensor mount. It's a huge
improvement but needs a screw adjust.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

Mjankor, 2019-09-05:
> @Mjankor,
> 
> It’lol come down to how well it can be incorporated in code but hardware wise it’s just one extra cable to the extruder and a slight redesign of the extruder to house both sensors. I’d probably still go direct to limit part count and resistance on the filament.

Yeah, extruder design is pretty well done (though I might just take a look at the Pinda Mount to see if it can be moved away from the hot zone a bit. I've been seeing a lot of reports of PETG extruders warping). 
https://www.thingiverse.com/thing:3431164

TheZeroBeast, 2019-09-06:
Please have a look at the latest release. Nice extruder design @Mjankor, check out my TOWER ELIMINATOR! referenced on the release page, it's such a great reliable solution to the silly tower.

closed 2019-09-06 by TheZeroBeast

AbeFM, 2019-09-06:
Heh, I was JUST drawing up a taller tower extruder cover. I have been using one for months and it is GREAT but the particular design needs a lot of work for adjustability and fit.

It seems like if you moved your pivot point down on the sensor flag you could increase your motion ratio? Do you have issues with, for instance, NinjaFlex and PLA in the same print? I have found the ~1:1 motion ratio setups all kinda get touchy when the deflection of the idler is <1 mm.