TheZeroBeast/TZB-MMU2S-Firmware #122: finda to extruder laser sensor

sotongster, 2019-09-04:
hello, i read with great interest that you managed to have configure finda probe together with extruder's laser sensor. 

Which file will i be able to see the changes at? Also, as such, there shouldn't be a need to calibrate bowden lengths right? i.e. mmu2 can be set to infinite filament loading until IR on extruder is triggered. 

Is my understanding correct?

TheZeroBeast, 2019-09-04:
Hey,

You can read up on my MK3:MMU2 changes here. It still required you to set the Bowden length on one channel for a ballpark so that the majority of the load could be done fast instead of loading slow the entire way waiting for it at the laser sensor.
https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases/tag/2.1.8

This will be the same approach I take for the MK3S:MMU2S, although now it’ll be going down to the IR sensor instead of the laser one.

TheZeroBeast, 2019-09-06:
Please have a look at the latest release, once stable I'll be looking into further improvements like possibly added the laser sensor back in to assist with Jam detection.

closed 2019-09-06 by TheZeroBeast