TheZeroBeast/TZB-MMU2S-Firmware #123: mk3s+MMU2

sotongster, 2019-09-07:
hello, i read the new release and seems like your latest mmu2 firmware is to work with MK3S's. 

If i understand correctly, prusa's MK3s firmware already allows the FINDA to feed directly to IR sensor at the top of the extruder. May i know what is the difference in your MMU2 firmware as well as your MK3s firmware? (sorry if i missed them. hope for your forgiveness and advice)

TheZeroBeast, 2019-09-08:
Hey, checkout the releases page for differences.
The best way to see the difference is load and see, you won't be disappointed.
https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases

closed 2019-09-08 by TheZeroBeast

AbeFM, 2019-09-08:
Probably tomorrow. Can't wait to try it. Developing new hardware and wanted to see it work once on stock firmware before I change.