TheZeroBeast/TZB-MMU2S-Firmware #129: Filament Tube Calibration Broken

AbeFM, 2019-09-10:
Every time I run the calibration routine, the selector is on Ch0, the idler cam is on Ch1, so the filament comes out NEXT to the tube.

I adjusted the length anyway, and when I press the middle button to 'save', it instead rehomes selector (would get caught on filament sticking a foot out of the machine but I have no blade and was using TPU), goes back to position CH0, then feeds out another few inches on CH1.

Note: The length that it feeds at first (on Ch1) is ~3", nowhere near the length of the tube, it takes about 8 adjustments to get it there.

This survives resets, and a power cycle.

I did it as described again, after power cycle, with ch1 having a foot of filament hanging out. When selector returned to position Ch0, this time it fed the right filament (Ch0) the right amount (~4 cm past stock tube length).

Also: When working right (Ch0) the steps are small (<~1cm) and instant. When feeding the wrong channel (as described above) it was rolling the idlers into position for each move, and each move was ~1-2".

Lastly: Update the instructions to say if it;s better to go slightly past or slight shy of the end of the tube since you're unlikely to get it exact.

Penultimately: OMG it's awesome, it;s so fast and smooth. I feel like my life effectively got longer by the 40 hours I'll save not watching it home and move slowly this year.

TheZeroBeast, 2019-09-11:
Note, never use TPU/TPE or any other flexible filament for tube length calibration.

I've made a lot of changes to make the setup menu and Bowden cal work as it had some bugs from old. Please re-review when RC2 comes out.

Yes, much time will be saved by all, exciting to have this almost working on the MK3S.

AbeFM, 2019-09-11:
Glad to see it!

Yes, I used PLA for the calibration. :-)

It seemed buggy, first few times it was weird, but then seemed to work fine.

Looking forward to next release.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes