TheZeroBeast/TZB-MMU2S-Firmware #130: Heated Bed Rapid Clicking

AbeFM, 2019-09-10:
Not sure how big an issue it is, but I hear the heated bed clicking rapidly (~5 hz?). It's not very loud but sitting near printer you definitely know it is happening.

I suspect it is louder than more recent stock firmware but it could be in my head. Not sure if there's meaningful data I can provide but if there's a test worth running let me know and I will.

TheZeroBeast, 2019-09-11:
There will always be an associated noise when switching at 30Hz and due to poor PRUSA/Marlin code that calculates or more accurately doesn't calculate PID correctly you'll have strange small full on and off oscillations. You can minimise this for now by doing a 70c bed PID cal and updating your PIDs. I plan to redo the bed PID in the near future.

The low 30Hz is due to poor EINSY Board design as the MOSFET switching the bed doesn't get turned on hard enough to switch any higher without causing issues.

closed 2019-09-11 by TheZeroBeast

AbeFM, 2019-09-11:
is it the same as Prusa's bed code?

I know some iterations they were wrecking hardware - I wanted to make sure
the clicking isn't that.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes