TheZeroBeast/TZB-MMU2S-Firmware #131: Loading during print doesn't work

AbeFM, 2019-09-10:
My first print with new FW seemed to load filament wrongly - not quite sure what's going on but it seems tangled up in extruder (door moves more than normal, and I only get dribble from the nozzle and clicking at extruder).

Using "load to nozzle" from the menu behaves as expected and loads without issue.

Also, the loading is VERY slow! It's slow till the end of the tube. Perhaps this is 'silent mode' but it's really slow.

Still digging into this. Ran it with the door off and haven't figured anything out yet, but the system's previous several hundred loads on OEM FW worked fine.

AbeFM, 2019-09-10:
It's possible this is from poorly executed tube length calibration. Still looking at it. I got it to fail on a 'load to nozzle'.

I fear I have a few mm more distance from end-of-tube to extruder gears. Given the differences in extruders, I'd suggest changing the calibration to be back to the gears - or at least express it in the ReadMe as calibrating XX mm before the gears.

AbeFM, 2019-09-10:
Sorry for the poorly explored bug reports.

By adjusting sensors I've had more luck. Load speed is much better out of Stealth mode. I will continue to compare loading in both modes later - right now I think I have it loading and am doing a brief test print.

AbeFM, 2019-09-10:
[image]

Not sure how this is happening. Somehow it wasn't loading one channel, but didn't notice and kept right on printing. I paused print, manually loaded once, then it worked and worked from then on for a few cycles until I stopped print.

The next print failed to load on first load. I'm not sure what I'm missing here.... But I'm having issues loading.

TheZeroBeast, 2019-09-11:
As per instruction you must have calibration to the end of Bowden or you'll get extruder grinding.
RC2 which is coming soon improves the load sequence as RC1 was a starting point. Please re-review when RC2 is released.

closed 2019-09-11 by TheZeroBeast

AbeFM, 2019-09-11:
"At the end of the tube" can be over- or under-estimated due to finite step size. And since since extruders have the tube end in differing locations relative to extruder gears I was looking for something more exact "ten mm from gears", or similar.