TheZeroBeast/TZB-MMU2S-Firmware #136: TPU loading WAY too slow

AbeFM, 2019-09-13:
The old speed worked just fine for me, without issue. The heatbreak jam thing is cool but so far only false triggers.

Out of ~30 load attempts, maybe 2 worked. I got a LOT of heatbreak jam detected, and a fair amount of filament sticking out the side of the extruder.

I went back to RC1 and things worked again. I feel the heatbreak jam detection is too aggresive, or the check (I assume it triggers when you see IR sensor turn off during load?) only for Type 1 and Type 3 filaments, but not for TPU.

TheZeroBeast, 2019-09-14:
what TPU hardness are you using?

I'll be trying hard TPU later today so I'll see how much I can adjust the speeds. My previous 87a TPU needed to be loaded that slow to give it time to expand down the Bowden tube. I'm thinking it's just too soft for a MMU setup though.

Jam detection I'll review once I try TPU, it's a basic 60step past bondtech, 52back and if it no longer has a hold of the filament there was something in the way in the heartbreak.

TPU may need an adjustment in this process though given it's so damn stretchy.

TheZeroBeast, 2019-09-14:
Hi @AbeFM,

I've done a few successful TPU print this afternoon, I did double the load/unload speed as it could be with the hardware TPU, I've also adjusted the load period for this type so it actually makes it to the bondtech gears, try loads now.

[TZB-TPU-Tweaks.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3612723/TZB-TPU-Tweaks.zip)

AbeFM, 2019-09-15:
I tried those, had nothing but failures. I always get heatbreak jam, in fact, it seemed to be more jam prone as I would occasionally have filament halfway down the heatbreak where JUST touching it would get it through.

I could "load to nozzle" without issue, repeatably, but when actually printing it was always failure.

Of note: The last time it got a jam, I opened up extruder, "move axis" to get extruder to turn, and completely fee the extruder wouldn't turn. It was making a high pitched squeal and my assumption is motor current was very low. I could overpower it with the thinnest of dental picks - one I would assume would bend if the motor had even half power.

Flashing firmware back and forth, I consistently have it work on RC1, and fail on RC2 of either flavor.

Probably unrelated, but see this discussion (seeing failures on load had me wondering if brief flicking of IR sensor during load was at fault) - https://forum.prusaprinters.org/forum/original-prusa-i3-mmu2s-mmu2-hardware-firmware-and-software-help/extruder-gears-out-of-round-tpu-ir-sensor-poor-performance

TheZeroBeast, 2019-09-16:
Monitor the sensor status page while you load. It should be solid off and then on once filaments in the teeth 🦷 of the bondtech or you’re not going to get a load.

Also, when the stepper gets disabled which is in all mmu error states, currents are dropped to 0 to allow you to pull filament out and so on.

AbeFM, 2019-09-16:
Good to know about the current dropping when in error state. Please reconsider request to maintain printer temperature for a minute (or even seconds) after a hang. Right now I still have to go on a LCD Menu Safari to get the nozzle heated to deal with these issues.

If you checked the thread where I linked, you'll see there are door positions which are wider when empty than the most extreme when it is empty. This is why it often fails checks where the sensor status is monitored while the door turns.

Check that thread, I'll have videos posted soon. I don't think your firmware can do a lot to make up for this, but it's the issue I'm facing.

TheZeroBeast, 2019-09-16:
I checked the thread and yes that does appear to be an issue when not setup to prevent it.
The code needs a solid sensor reading to do anything intelligent with it.
If you’re running the extruder and you’re getting any false positive IR_SENSOR readings please review hardware setup.
Might I suggest reviewing my suggested design as it appears I have fixed that issue without being appear there was one.
Having never given the silly tower a chance 🤦‍♂️

P.S. I got 30 loads into a two TPU print with current code which is very promising. Doing more tweaks to improve it’s handing👏
[image]

TheZeroBeast, 2019-09-17:
I just  rampped Flex speeds and accelerations up further in attached build.
[TZB TPU Tweaks 2.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3620986/TZB.TPU.Tweaks.2.zip)

I use these Filament setting with Flex with the new speeds in attached Hex files....
[image]

AbeFM, 2019-09-17:
Uh huh, I've used 3 or 4 different extruders, and a bunch of firmware. There's not a lot you can do with a digital signal that's unreliable unless you get really fancy with repeated reads and knowing where you are, etc etc.

The practical side is that I have had it working, I've done several prints are various hardware and firmware with >400 changes... It just seems to be a matter of getting the sensor perfect, and it's so hard to tell when it's perfect enough that it won't shift a few microns during the print.
[image]

As to fixing it, I just don't know. The issue comes down to there being a 600 um gap to start with, another ~150-200 material sunk between teeth, so now you're trying to read 1750 - 600 - 200 = 950 microns at best. Since the material squishes some, then suddenly the 200 um of error becomes an issue.

That's a very low volumeteric speed! I think I'm using ~5, but I'm using the Shore 98a stuff.

AbeFM, 2019-09-17:
I don't want to pollute this software development with physical stuff, but I would appreciate if you could read your door deflection as you run the extruder empty.

TheZeroBeast, 2019-09-18:
I did a IR_SENSOR test this morning @AbeFM, no false positives at all, this is even with TPU so it's definitely not impossible to have it tuned and working.
[image]

AbeFM, 2019-09-18:
Can you video the door to see how much it moves while the motor feeds? Maybe go all out and put a ruler or piece of filament near it for scale.

The video might have that already, but it's set to private.

I can find a range where it works, but only barely. Yesterday's test of Tweaks2.zip ran all 100 changes with no full stops, only one 'mmu load error', the one before (~60 microns of adjustment) was 5 mmu errors, no load errors.

The concern is if your hardware is unusually good, will it work on the average machine? 

Anyway, happy to have it functioning again. **The faster loads helped**. I'll check slicer settings against yours.

It would be pretty awesome if you could find a way to use the 'max volumeteric print rate' or something else intelligent to get the feed rate from, since "flex" really is not enough. They need a HARD and SOFT flex, imo.

TheZeroBeast, 2019-09-19:
Video fixed 🤦‍♂️

I'll see what I can do with a ruler in a video.

If stock design and software is flawed people may have to modify to get a stable setup.

Interesting but really soft is just out of the question in my opinion.

closed 2019-09-19 by TheZeroBeast