TheZeroBeast/TZB-MMU2S-Firmware #137: Enhancement: Preheat to Last Used Temperatures

AbeFM, 2019-09-13:
When Pause Print is selected, the printer sets the nozzle temperature to 0.

Most of the time Pause is selected, it is to work on something. Since the head lifts, I suggest there is no harm in leaving the nozzle heated for ~1-2 minutes, if there are no interactions.

Also:
I'm printing at, for instance, 220/50. I have a plethora of options to preheat to, but as none of them are the temps I'm using, I have to wait for a temperature change, load, unpause, then wait for it to go to the print temperature from the preset one.
If "Temperature When Paused" was an option it would be a LOT more useful. 

At minimum, it is inconvenient. For some materials, it would ruin a print.

TheZeroBeast, 2019-09-14:
Sorry @AbeFM, I'm not got to update the preheat options although you're welcome to adjust them yourself and rebuild, easy as.

As for the pause routine I'm also not interested in changing it from stock so feel free to fork and make it how you want 👍

AbeFM, 2019-09-18:
Response made more sense here than in Comms issue:

Most of the time I'm having an issue, the printer doesn't seem to know it's having one. Sometimes it sits there for minutes, with nothing anywhere moving..... Then just goes again.

I'm saying it's not always in an error state when I reset it. Probably I have to take more care, but since I've had days where 95% of the loads don't work (like heatbreak jam) I find myself spending a LOT of time setting temperature, wait for it to go to unrelated temperature, , unloading, trying to load, setting temperature again, wait for it to go to unrelated temperature, having it load the wrong filament, unloading, setting temp again, wait for it to go to unrelated temperature, hitting resume, it just sits on prime block, pause again, set temperature again, wait for it to go to unrelated temperature, etc etc.

The bigger complaint is how hard it is to deal with the printer trying to put it into a reasonable state. The extra steps make things like paying attention impossible because you're spending all day just trying to keep the nozzle heated.

Not terribly useful post, I get it. But the hard to deal with error states are a show stopper in terms of doing any actual evaluations.

TheZeroBeast, 2019-09-19:
Sounds like you're using RC1 if you're getting pauses with no error state mentioned.

Like I said above, fork, make your changes and pull-request back in for review.
With the error states and action commands in Octoprint it is a million times more troubleshooting info to work on than stock so winning.

AbeFM, 2019-09-19:
Running octoprint would be a good idea. I'd have to set it up though. Too many people complain about print quality and pauses when printing over USB for me to really make that a selection - I don't know if it's because of constant requests for data or just bad serial, but I'd like to set the printer up to just accept an SD card and spit out plastic items. :-)

https://photos.app.goo.gl/jDyYhhgMHaCJsiD58
I've seen this on RC1, 2, and the couple firmwares you've posted as zip files here. This was on the latest one you posted.

I believe it to be the MMU and the mk3 fighting for the display, but it's hard to say.

AbeFM, 2019-09-28:
Have PVA in extruder.
Pause print
Select "unload filament"
Only choices are 215c or above. All choices will destroy a nozzle.

Consider adding a "last used temp" option.

TheZeroBeast, 2019-09-30:
We should check if PRUSA have this pegged as it really needs to be in stock.

TheZeroBeast, 2019-09-30:
If not I think it's an easy add to the preheat list

AbeFM, 2019-10-01:
The pretend PVA doesn't exist because after they said you could print it, they changed their minds.

It's a full disassembly if you bake stuff into your nozzle, and likely new parts.

My suggestion about last used temperature (which I believe IS stored in memory) was that anyone can use any material and it is instantly supported. Trying to make a list with every number on it just makes for a list so long no one will read it.

Otherwise, a preheat option that is nozzle only (since bed is held at last used temp anyway) makes way more sense than changing the bed temp. It's just a BAD menu - every single option on the list will cause a problem, the advantage being choosing by name is more user friendly than having to choose by number?

The less it jams the less of an issue it is, but still. Glad it's on the list of enhancements.

TheZeroBeast, 2019-10-27:
Hi @AbeFM,

I'm looking at this now.

The solution I've come up with is a preheat option at the top that appears once the extruder has had at least one set temperature.
As no previous heater set temperatures are stored(once they're set to '0' you have no reference to previous temp), I created a previous variable to store the extruder temp whenever it's set to something not '0'.
This variable is set when the new preheat menu item is selected.

Due to how that menu is made I haven't been able to include what it'll set to when selected, maybe a code guru out there could suggest a solution to this.

TheZeroBeast, 2019-11-04:
I have been running it for a week now and I find it helpful, not ideal not knowing the temp you're going to, I'll see is another code guru can assist with a fix.

Give it a whirl and let me know what you think @AbeFM 

[TZB-MK3S-3.0.3-ALPHA-BN2474.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3802476/TZB-MK3S-3.0.3-ALPHA-BN2474.hex.zip)

AbeFM, 2019-11-05:
Running BN2476, which I assume has the same changes in it - and just giving it one quick test from what happened to be sitting on the SD card and happened to be loaded into the machine, yes!

It seems to do exactly what I want, and it IS super helpful!

If I do a more complex, multi-material print and I see it misbehave, I'll update. Thanks again, this was a minor but constant pain point for me before. It'd sure rock if PR would roll some of this fine work of yours back in. :-)

TheZeroBeast, 2019-11-06:
Glad it's as intended, I'm still not fully happy I can't get the value to display, as mentioned above I am seeking some further assistance and will update on any progress.

closed 2019-11-06 by TheZeroBeast

AbeFM, 2019-11-06:
I agree it would be nicer looking, but having it working is the big
difference. Will save a lot of time!

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

AbeFM, 2019-11-06:
Not sure if this is a related bug, but I haven't seen it before:

Start print.
Notice your filament feed is garbage, pause print and fix it using awesome Preheat to Last feature.
When you select 'Resume print', I get:
[image]

I double checked I had "fan check" off (using a 2 wire fan, ironically while printing an adapter for a 3 wire fan). I tried it with fan check on, and off. Note: The layer fan pulses briefly at the same time the warning comes up, presumably testing it as well.

So I think you need to respect the "fan check" variable when resuming print. Unsure if it's related to the Ext Temp changes, but I never had that happen before that I can recall.

Note that it started the print fine, and without a reset started the next print fine so I suspect it only happens coming back from pause.

TheZeroBeast, 2019-11-07:
Yeah, that's not looking good. Can you please open an issue for this on the PrusaFirmware repo?

Interesting, I bet if I tried and still had a 3 wire fan attached it'd read the rpm and continue 🤔
I'll try find in my codebase to fix also, I'd say you're correct about the resume from pause.

AbeFM, 2019-11-07:
Your mk3 repo or the official Prusa one?

I.'m not sure it's in their code yet...

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

TheZeroBeast, 2019-11-07:
I'f you haven't changed your fan over yet would you please do a test with stock MK3S fw to see?

AbeFM, 2019-11-09:
I would want to know the loading-distance (gear to nozzle distance) for
your extruder so I could compile it. :-)

AbeFM, 2019-11-12:
If you want to provide that information, I can find the stock firmware, download it, compile it, upload it and report back. I suspect you have the capability to do that much faster yourself - but I would be willing to do it if you could give me that number/variable so I can do the rest.

Getting that number would be enabling so I could "please do a test with stock MK3S fw to see?"

AbeFM, 2019-11-18:
https://github.com/prusa3d/Prusa-Firmware/issues/2331

Looks like there's a fix in the works.