TheZeroBeast/TZB-MMU2S-Firmware #138: Original Prusa MMU 2.0  control device could not be found

iflawdya, 2019-09-14:
Had TZB RC1 installed with no problems. Just updated to RC2 starting with the MK3s first. No problems. Went to do the MMU2s and got the error Original Prusa MMU 2.0 control device could not be found. When I bring up the configuration tool and plug in the usb to mmu it scans and Arduino Leonardo is what comes up in serial port.

TheZeroBeast, 2019-09-15:
Hey,

Try re-download RC2 and always ensure you’ve just reset the mmu2s before flashing as it has a 5s flashing window.

iflawdya, 2019-09-15:
Strange but it worked. I had to reset as I was flashing firmware for mmu2s.

TheZeroBeast, 2019-09-15:
Yeah, just how they’ve setup the bootloader. Reset MK3S also resets MMU2S btw. If you were resetting the mmu2s itself

closed 2019-09-15 by TheZeroBeast