TheZeroBeast/TZB-MMU2S-Firmware #140: Startup Error

AbeFM, 2019-09-16:
RC1, log when turned on connected to S3D:

`SENT: M105
READ: ok T:26.5 /0.0 B:25.7 /0.0 T0:26.5 /0.0 @:0 B@:0 P:26.8 A:35.2
WARNING: Device unplugged while connected to port
Disconnected.
Attempting connection at ...
  Error opening selected port.
You must be connected before sending commands.
Attempting connection at ...
  Error opening selected port.
Attempting connection at ...
  Error opening selected port.
Attempting connection at \\.\COM4...
READ: start
  Connected to machine!
SENT: T0
READ: echo: TZB3.0.0-2427
TZB3.0.0-2427
READ: echo: Last Updated: Sep  7 2019 16:03:19 | Author: (none, default config)
Last Updated: Sep  7 2019 16:03:19 | Author: (none, default config)
READ: Compiled: Sep  7 2019
READ: echo: Free Memory: 1933  PlannerBufferBytes: 1392
Free Memory: 1933  PlannerBufferBytes: 1392
READ: echo:Stored settings retrieved
Stored settings retrieved
READ: adc_init
READ: CrashDetect ENABLED!
READ: FSensor DISABLED
READ: 
READ: echo:SD card ok
SD card ok
SENT: M105
READ: echo:Unknown command: "0"(2)
Unknown command: "0"(2)
READ: ok
READ: ok T:26.8 /0.0 B:25.9 /0.0 T0:26.8 /0.0 @:0 B@:0 P:132.8 A:33.0
SENT: M105
READ: Command not found!
READ: ok T:26.7 /0.0 B:25.8 /0.0 T0:26.7 /0.0 @:0 B@:0 P:61.7 A:33.1
SENT: M105
READ: ok T:26.7 /0.0 B:25.8 /0.0 T0:26.7 /0.0 @:0 B@:0 P:30.1 A:33.5
SENT: M105
READ: ok T:26.7 /0.0 B:25.7 /0.0 T0:26.7 /0.0 @:0 B@:0 P:27.1 A:33.1
READ: MMU => MK3 '300'
READ: MMU => MK3 '320'
READ: MMU version valid
READ: MMU => MK3 'Active Extruder:1'
READ: MMU => MK3 'FINDA Initialised'
READ: MMU - ENABLED
SENT: M105
READ: ok T:26.7 /0.0 B:25.9 /0.0 T0:26.7 /0.0 @:0 B@:0 P:26.8 A:33.1
SENT: M105
READ: ok T:26.7 /0.0 B:25.7 /0.0 T0:26.7 /0.0 @:0 B@:0 P:26.8 A:33.5
SENT: M105`

Specifically this bit:
SD card ok
SENT: M105
READ: echo:Unknown command: "0"(2)
Unknown command: "0"(2)
READ: ok
READ: ok T:26.8 /0.0 B:25.9 /0.0 T0:26.8 /0.0 @:0 B@:0 P:132.8 A:33.0
SENT: M105
READ: Command not found!

AbeFM, 2019-09-16:
Oh, that might be PART of a line that I had sitting in 'send' which got poorly sent? "02" is in the middle of the "M302 S0" I'd tried to send earlier.

It is happening every time, though.

TheZeroBeast, 2019-09-16:
Not current build, has already been resolved.
Please flash RC2.

closed 2019-09-16 by TheZeroBeast