TheZeroBeast/TZB-MMU2S-Firmware #141: Improve Tube Calibration

AbeFM, 2019-09-18:
The 'end of the tube' calibration doesn't make sense. Not everyone's tube is in the same place. At the start of the gears is a lot easier to get right, I feel.

My current tube ends ~6mm short of the grab point, maybe less. So I tried to get it at the top of the tube connector, and that was too far. At least put in the documentation how far short of the gears you're supposed to set it so users don't have to do random trial-and-error as I did.

Also, I am pretty sure a well calibrated IR sensor is completely capable of setting itself by feeding the PLA a few times into the extruder and looking for when it catches/triggers, then you don't need to disassemble or manually figure out anything.

You've got feeding detection all over the place, I think it'd work. :-)

TheZeroBeast, 2019-09-19:
There is a 50mm sense window past the calibration point, perfect sense.

How far is your end of tube from your gears?

Auto is a possibility although not as simple and clean as it relies entirely on a solid IR_SENSOR reading which apparently isn't common with STOCK Hardware.

AbeFM, 2019-09-19:
Today it's whatever your build does, it appears to be very close to the gears. I believe it was about 35 in the past, so I'm not sure why it hadn't work until I calibrated it with a CM-and-a-half out of the tube.

Listing how far from the gears you'd like to be in the readme/calibration instructions would make it a LOT easier for people to know what number you're using. No programming required.

Similarly, if you DID have it tuned at the gears as Prusa does it, you could say "or, for stock hardware, calibrate it so 3 cm (or whatever your number is) past the end of the tube.

TheZeroBeast, 2019-09-20:
Ok, maybe to the gears is a better option 🤔 for two reasons.
 1. People already know PRUSA tube cal procedure.
 2. It then doesn't matter if tube is mounted higher as you experienced.

Change will be in RC3

closed 2019-09-20 by TheZeroBeast