TheZeroBeast/TZB-MMU2S-Firmware #142: MMU v2 filament load issue

Penfold81, 2019-09-19:
Hey. I have flashed the new versions of this zero beast and no matter what i do i cannot get filament 5 to load? 1-4 are fine but 5 just goes to the extruder and it faults out for load failure and will flash constantly. If i push the right button the filament unloads. Green flashing light, then i puh the middle button and it reloads but again goes past the ir sensor but will not load. it faults out before it gets a chance. only happens on filament 5

TheZeroBeast, 2019-09-20:
Hi @Penfold81,

Your fifth filament isn't making it to the gears, have you completed the mandatory Bowden tube length calibration and if so what extruder body are you using, ie. the STOCK tower of pain from PRUSA?

TheZeroBeast, 2019-09-20:
Hi @Penfold81,

RC3 is out which changes tube calibration back to the STOCK (center of bondtech)
Please load, re-cal to center of Bondtech gears and retry your filament 5 load, should be perfect now.

CaiJonas, 2019-09-25:
Hi @TheZeroBeast 
Very good work so far. Is it possible to get your extruder parts as a step file? Ofcourse I would donate a small amount.

Best Regards

TheZeroBeast, 2019-09-25:
Hi @CaiJonas,

I may consider sharing the step files.
What changes do you have in mind, I might be able to assist?

CaiJonas, 2019-09-25:
@TheZeroBeast I want to change the pin which sits on the lever and add an screw to the door to adjust the pin lenght. Also I have some problems with printing flex filaments. I often gets clogging error. Maybe the solution is to adjust some tolerances in your design system. But over all I am very impressed by your work. Without your ideas and solutions the mmu2 is not very good

AbeFM, 2019-09-25:
[image]

[image]

Use a set screw - regular screws have uneven tips. It works GREAT, even at M3 which I thought too coarse. About 1/4 turn is the overall range of adjustment needed (~150 um). I just drilled a hole and I don't remember if I ran a tap through, but I think I did. Certainly punching a hole in the STL makes sense.

I have a LOT of issues with the flag. TZB uses the same flag as stock, but it's not like PR never does shitty engineering. :-) I drew up a few - STEP WOULD HAVE BEEN EASIER and saved me many hours - and ended up trimming with razor, and putting foil over it too.
https://photos.app.goo.gl/f4ndFa96DneiCU737
I would STRONGLY suggest you check this isn't happening to you.

I can say that after much screwing with, I've got it working pretty reliably.

TheZeroBeast, 2019-09-25:
I'll have a look at adding a grub screw into the design.
The Jam error when using flex if a false positive would be the tune on the pin and idler tension which the grub screw would assist with.

AbeFM, 2019-09-25:
I can say from experience that the on-off-on of the IR sensor WILL cause a jam detection to trigger, even when everything else works perfect.

TheZeroBeast, 2019-09-26:
[TZBMK3S-Door.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3659561/TZBMK3S-Door.zip)

I'm thinking a 3mm hole inline with the PIN will be tight enough for a M3 grub screw and is optional (ie. it'll work fine without it, but give you the option to chuck it in if needed)

I've also included the step of the door if you'd like to try something different.

AbeFM, 2019-09-26:
Er, probably you want a 2.5 mm hole (that's the drill size for an M3x0.5 tap).

I'm unsure if a SMALL (1 layer) chamfer on the top of the hole would help keep the surface pristine for those who aren't using the screw.

Those who are likely want ~2 layers off the door so you are sure there's enough clearance?

CaiJonas, 2019-09-27:
Hi guys, thank you for all the suggestions. I will add a grub screw maybe @TheZeroBeast  will add a 2.5mm hole at the correct position and update the stl files on thingiverse. Additioning to that I will check correct sensor reading like @AbeFM. I can't remeber wether I changed the heatbreak or not, I have read somewhere this could also be a problem.

AbeFM, 2019-09-27:
Heatbreak is a good idea-  every time I touch it, stringing, etc, changes.

You can just drill a hole, or use 3D builder to create a 2.5 mm cylinder, and remove it. Put it in the middle of the flat spot.

TheZeroBeast, 2019-09-27:
All done and live.
https://www.prusaprinters.org/prints/5756-tzb-mk3s-mmu2s-extruder-no-tower

TheZeroBeast, 2019-10-02:
Stale, no update from originator. Please re-open if you're still having load issues with the latest release.

closed 2019-10-02 by TheZeroBeast

AbeFM, 2019-11-05:
FYI link goes to a 404.

TheZeroBeast, 2019-11-06:
Fixed here, I think I had to recreate my part for some reason 🤦‍♂️

https://www.prusaprinters.org/prints/7690-tzb-mk3smk25smmu2s-extruder-mmu2s-no-tower-v2