TheZeroBeast/TZB-MMU2S-Firmware #143: Screen Acting Badly

AbeFM, 2019-09-19:
Generally this is in an error state.

https://photos.app.goo.gl/jDyYhhgMHaCJsiD58
I've seen this on RC1, 2, and the couple firmwares you've posted as zip files here. This was on the latest one you posted.

I believe it to be the MMU and the mk3 fighting for the display, but it's hard to say.

TheZeroBeast, 2019-09-20:
Which error state specifically, there would be a logic Bug I need to fix and haven't found yet and you don't provide enough detail to narrow it down.

TheZeroBeast, 2019-09-20:
Please review RC3 and advise if issue persists.

closed 2019-09-20 by TheZeroBeast

AbeFM, 2019-09-20:
I see it often, often enough I'm not convinced it's just one state.

I will load RC3 and hopefully things are mostly working so it's easier to sort out the causes.

AbeFM, 2019-09-20:
It's REALLY hard to keep track of what I've reported with everything closed.

I do not consider an option with no fix to be closed. It's hard to find them, so duplicated, disorganized issues get opened.

I would suggest leaving unresolved issues open - it's not a question of if you've read it, but if the source of the issue has been found and testing has shown the issue to be resolved.

I feel like I spend most of my testing time trying to find topics and just shotgunning out anything that doesn't work because it won't get attention after the issue is closed.

Could be a misunderstanding of GitHub etiquette, and the project is awesome and you're reading this stuff and even knocking out issues - but I don't always know you're doing it, I don't know there's been a fix until I read the next release notes. Not telling you how to run your show but I'm going to run out of steam on this if I can't even tell what's being read. Most of the upgrades are awesome, btw.

TheZeroBeast, 2019-09-20:
1. You need to provide more information, I've even created issue templates for people to follow.
 2. If like a lot of the time it is from a previous release it gets closed.
   a. If you find an issue in a previous release then please see if it's still there in the latest release and as per the template provide all steps taken to get to that point so it can be reproduced and fixed.
 3. If it is clear what your issue is and it gets fixed, a version/build number will be provided at time of closing issue.

Take this issue for instance that you opened this morning not using the template (when you could have), no steps on how you got to the displayed error state.

I appreciate your help but it's not productive if you just show me something wrong, tens of thousands of lines of code could hide your bug you see.