TheZeroBeast/TZB-MMU2S-Firmware #144: More screen blanking, removing filament before MMU boot

AbeFM, 2019-09-20:
Filament fails to load, printer stays "locked", screen goes blank, occasionally lights for fraction of second as you interact with it.

Reset machine. When menu appears, select "unload filament". Instead of unloading, it pops up menu "which filament to load?".

In other circumstances, a failed load makes temperature unset, as machine keeps making attempts to load/unload, temperature drops. My last unload happened at 165* C, which is way too low for the TPU that was in there.

Generally states are not being kept track of enough. All the threads I open end up closed so fast I can't keep track of what I've reported or haven't, what's been fixed or what hasn't. I can't really give more detailed information without more interaction - all I can do is post "it didn't work again".

Going to run stock FW for a bit and see if machine is reliable before I try this again - the screen doesn't work, loading doesn't work, unloading doesn't work, I can't adjust anything, it just doesn't handle errors elegantly enough, my hand is floating over power button because it keeps doing bad things.

I think part of the reason is that the IR doesn't turn off all the time. With PLA loaded, and the door moved >1mm, the LED stays off.
If I hit the extruder with a small mass, I can jar whatever's stuck and the IR sensor will trigger.

I couldn't get it to work without a piece of tape stuck to the inside of the door (in an attempt to get what would be 1/4 turn on an adjuster screw), I used alum tape so it would be smooth but it may not be appropriate. Possible a set of adjuster arms, ~15 of them each longer by 10 um might let me easily adjust this, but if something shifts or relaxes slightly I guess it'd be another day of swapping posts?

--------------
Load to nozzle works >10 times in a row.
Loading during 'live Z" fails with filament sitting above gears.
Pressing buttons on MMU makes it pull filament back up, and try again.
Mk3 doesn't run extruder gears on subsequent loads.
Selecting "load to nozzle" on that screen (stuck during print) makes screen blank.

Without physical changes, all loads ok from base menu, all printing loads fail. Light doesn't light and it looks like not enough filament is fed. I'm using a loose tube (3mm ID) from MMU-extruder, perhaps there's too much room and it's giving up? What distance from the gears would you suggest I calibrate for? I can run a tighter tube, if it helps. I like that you're not feeding too much filament on a failed load - but I'd like the loads to work. :-)

AbeFM, 2019-09-20:
The poor loading belongs in another thread - however it got much better when I calibrated the distance closer to the gears.
Based on my experience, I would recommend that the calibration be done such that you're 15 mm from the grab point. If you do that, loading in prints works, setting it >25 does not.

TheZeroBeast, 2019-09-20:
As per RC3 cal to center of Bondtech, problems should be away.
If you post it didn't work it won't get fixed.

Please use the template and give all the requested information.

If I can't reproduce it can't be fixed.

closed 2019-09-20 by TheZeroBeast