TheZeroBeast/TZB-MMU2S-Firmware #146: With 'filament sensor' enabled, unloads at the start of every print

AbeFM, 2019-09-20:
Set filament sensor to on (spool join turns on, too), start a print.
Before MBL, but after home, printer unloads even though it isn't loaded, and sensors all read 0.

Machine presents filament and cools down. Even an immediate response leaves the printer needing to reheat to do it's actual load.
When the printer sits, reading 'waiting for preheat', >5 degrees before setpoint, printer loads anyway, and finally gets to temperature after load.

MBL is done with filament loaded, and print begins, but initial purge is skipped (first bit of purge line) so thin part of purge line is just a row of random dots, but my purge tower outline and brim printed ok.

TheZeroBeast, 2019-09-20:
Please report using template so I have information to assist, you don't even state what version you're running which at time of reporting needs to be RC3.

closed 2019-09-20 by TheZeroBeast

AbeFM, 2019-09-20:
Wow. Rough. 13 minutes ago is 6:56.

I got the notice at 6:45.

That's barely enough time to load the firmware if all I did was sit there hitting refresh on my email and had the printer already plugged in and no testing going on.

I'm not even sure how to describe the version I got, it wasn't a release, it was a zip file I got sent, in a pile of other zip files.

Like my other comments, with threads closing down so fast I can't keep track of what I've posted, it's a lot of work to fill out the template. I will do it from here on out, but if every thread ends up closed when I'm still having issues then I'll be out of the beta testing business. I would really like to help. I'll try out RC3 when I get a chance and let you know - but if it's upsetting I'm 11 minutes behind then I just can't work on this project.

If you thought I didn't give you enough information, then ask for more. If there IS no issue and I was mistaken, how would you know that if I hadn't given you enough information?

I'm going to set this aside till tomorrow. I appreciate what you're doing and hope to be able to assist in the future.

TheZeroBeast, 2019-09-20:
Use template man, simple.
BN is a part of version and is unique....

AbeFM, 2019-09-20:
I will use it going forward. But if I'm testing with a 12 hour print and you're going to delete without consideration any report I make for a previous version then I am wasting my time. My overall point was with the rate these threads close there's no point to putting in the extra five minutes to fill out the form. Stuff closes - some of which you've fixed and I don't know about, and some of which I can't tell why. Some stuff isn't yet a reproducible bug (I've mentioned - most of this is fairly rare and spread out over 12 hours and inconsistent) but I recognize it as undesired behavior and I need a place to communicate about what I'm seeing to make sure I'm not doing something wrong, etc.

Going forward I will use the template but if a bunch of issues gets closed without being considered I'm going to end up back where I was lately - throwing stuff over the fence with 30 seconds put into it.

BTW - if there was a 'offline' way to communicate for stuff that doesn't rise to the level of bug-report, I wouldn't have to make one. If you have email/chat/etc that would be better for discussion (like the length of the pole on the extruder) that doesn't really belong as an "issue" here.

You're right on the BN thing. I was just feeling like these were getting closed regardless of content when in fact much of it was getting fixed. I'm not going to harp on that, I only have a week till I'm away from printers for a month, so I really want to get what progress I can in by then.

TheZeroBeast, 2019-09-21:
Thank you, it'll definitely help find those damn bugs and yes some can be hard to nail down when printing unattended. BTW, I just completed an 18h twin TPU print without any intervention on RC3, ~230 changes.

Feel free to you facebook messenger for general discussion, "Robert McKenzie" should be how to find me.

AbeFM, 2019-09-22:
RC3 (2444/352), after a lot of messing around to get stuff sorted mechanically.... unloads even when empty on the beginning of every print - with sensors reading correctly.

Afterwards, it loads, MBL, then tries to put too much in extruder and you're not extruding, only clicking.

That's with loading working great from the menu's "load to nozzle".

Flashed 9/7 prusa firmware (3.8.0/1.06) it does the same thing. I know this is something they fixed before, looks like it came back.

TheZeroBeast, 2019-09-22:
Wouldn't be printer profile not stock?
This would explain menu working and gcode not.

Stock for MMU2S is as bellow.

[image]

AbeFM, 2019-09-22:
I just keep getting filament bunching up in the cooling tube, I just swapped the PTFE (old one was 1 mm to short, perhaps this is too long....).

This is supposed to be a report about the filament being unloaded when 'filament sensor' is on in settings. Does yours do this?

AbeFM, 2019-09-22:
> Feel free to you facebook messenger for general discussion, "Robert McKenzie" should be how to find me.

Found like 6, none were obviously you. :-) Email address you use on there, maybe?

TheZeroBeast, 2019-09-22:
I'll double check if mine does that when the current test print is done, normally filament sensor should be on in the MK3S menu for the IR_SENSOR to work.

Robert.M.McKenzie@icloud.com is the one

AbeFM, 2019-09-22:
This is odd. On both your firmware, and stock, I'm getting fine loading, and fine printing... EXCEPT the first layer, first material is super underextruded. The beging of prime line is ok, but the entire first layer (prime block and brim and support) is massively underextruded, with extruder clicking most of the time.

There's a good gap, I don't think it's z-adjust. And the second material (and the rest of the print) seems to be ok.

This is new and I'm not sure what's going on. It's not really your firmware but I'm stumped anyway. Any ideas at all on where to start looking?

AbeFM, 2019-09-22:
Maybe I need to wipe out the printer's storage. Everything else is great except it's jammed up on the first load. Doesn't matter which channel it starts on. And it does kink the filament, I don't think it's that the current is low.... Though it never seems to shove out too much , it seems like there's something wrong with the first load.

[image]
2nd and 3rd material are fine....

AbeFM, 2019-09-22:
It's definitely kinking up the filament on load. I can pause print, find a nub at the top of extruder gears, bunched up at entrance to cooling tube. Perhaps just a chamfer on the inlet would help.

Anyway, if I unkink it after that load, it continues on ok, printing the layer fine. I still seem to get this sometimes, almost as if it's loading a bit too much.

Also, I don't get that super thick first bit of prime line - instead I get nothing, followed by a normal line transitioning into the underextrusion shown by the time it traverses the plate.

AbeFM, 2019-09-22:
I'm getting frequent jams when trying to unload after 'stop print' command. It almost always ties around driven extruder gear. Is it possible MMU isn't "sucking" up as much as it being ejected?

I've gone through a few PTFE's, still getting unprecedented issues on loading. Will clean nozzle tomorrow (already half cleaned a couple times today). Not too sure what else to look at - things look pretty solid, otherwise, I think the sensors are all cleaned up and behaving, just getting jams. I've had my PVA jam a couple times after printing some ok.

Bed time. :-)

TheZeroBeast, 2019-09-22:
Sure sounds like a PTFE issue in your extruder, if you don't have the chamfer then you're filament will most likely be jamming on the edge.

Get the length for the TZB extruder then cut the chamfer in and try again.
Especially important for TPU as there must be nothing to catch the filament on it's way to the nozzle.

https://manual.prusa3d.com/Guide/How+to+trim+PTFE+tube+-+Original+Prusa+i3+printers/500

[image]

AbeFM, 2019-09-22:
I found baked on chunks in my heat break (thanks keyboard for auto correcting to heart break), I suspect removing then will help a lot.

I'm very confident in my PTFE's, I only brought them up because your design doesn't fit the stock tube. Might be worth putting in notes of it isn't.

Overall this looks really promising. I've had months where I could bang out prints in any material I wanted here and there, I feel I'm nearly back.

TheZeroBeast, 2019-09-25:
[RC4.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3652282/RC4.zip)