TheZeroBeast/TZB-MMU2S-Firmware #147: Bondtech BMG Extruder

gizzburn, 2019-09-22:
Hi Robert I have decided to go with  the bear like X carriage with my Bondtech extruder, I know about the e-steps but what about the below statement from a post on FB. 

<Ah the z max pos will still need to be modified to 204. Not sure how to do that without modifying the firmware (which I don’t know how to do!)>
This is a quote from the Bondtech Site.

bondtech_fw_MK3S_3_7_2.hex for MK3s

2019-07-12

Compiled firmware to support Bondtech Prusa i3 MK3s.
Bondtech specific changes

    Estep set to 830.
    Z_MAX_POS 204 to allow Z and XYZ calibration.
    Changed printer name.

Will it need a patch for this, or can we do it like the e-steps way.?  Or can you port a version for the MK3s & MMU2s with the BT BMG Extruder..

Mick

AbeFM, 2019-09-22:
If you're still in development, it might be faster to just use the g-code
command to set the number you want.

https://reprap.org/wiki/G-code#M92:_Set_axis_steps_per_unit

Don't forget to save eeprom after so it sticks.

TheZeroBeast, 2019-09-22:
For now just use the following two serial commands once firmware is flashed to update the e-steps.
Keep in mind not to go over 200mm and you're set.
Once I'm ready for a full release I can add a hex for the BMG Extruder.
M92 E830
M500

gizzburn, 2019-09-22:
Thanks Robert, I have loaded the Bondtech firmware for now that includes the codes, will wait now till you release the full version for the Bondtech, not got any mulit color prints planned as yet so will make do till you get this done, great work as usual thanks.

AbeFM, 2019-09-23:
This isn't the first thread to have concerns about overall height. Especially since if the MBL doens't go well, the printer runs head up to the top (sometimes I had to watch out for interactions between chimney and MMU)... I don't know if there's some number you can similarly change with serial commands to fix that.... If so, the two (overall z height and esteps) could make it into readme. :-)

TheZeroBeast, 2019-09-23:
There is no serial command to update max_z_height sadly.

gizzburn, 2019-09-23:
I have messed my printer up, in the eprom I have typed in the gcode for esteps and the z one, but I think I have messed it up, now when I try to calibrate Z the motor moves slightly but does nothing, how do I reset the eprom, tried a factory reset but that did not work. Even tried other firmware but nothing changes.

TheZeroBeast, 2019-09-26:
If you've loaded stock FW and reset to factory and it still doesn't work I'd try run that gcode to set esteps back to stock values (just to make sure as factory reset should be doing this)

TheZeroBeast, 2019-10-02:
@gizzburn, the bmg, are they the same changes that Bondtech do for the MK3S at their FW link?
https://github.com/BondtechAB/Bondtech_Prusa_i3/tree/master/Firmware/3.8.0

gizzburn, 2019-10-03:
Great Robert long print on at the moment but when finished will give it a try many thanks.

TheZeroBeast, 2019-10-03:
Same MK3S changes as the link for the BMG mate?

TheZeroBeast, 2019-10-04:
Release page now has Bondtech’s changes in an additional HEX file as per the MK3S changes [here](https://github.com/BondtechAB/Bondtech_Prusa_i3/tree/master/Firmware/3.8.0)

closed 2019-10-04 by TheZeroBeast