TheZeroBeast/TZB-MMU2S-Firmware #148: MMU not recognized by PC when "TZB-MK2.5S-3.0.0-BN2450" loaded in MK2.5S

habitforming11, 2019-09-23:
**TZB-MMU2S Release Version**
TZB-MMU2S-3.0.0-RC3-BN352

**Describe the bug**
With MK2.5S firmware "TZB-MK2.5S-3.0.0-BN2450" the MMU2S unit is not recognized by my Windows 10 PC.  As such, MMU firmware cannot be flashed to TZB version.

**To Reproduce**
Steps to reproduce the behavior:
    Load "TZB-MK2.5S-3.0.0-BN2450" to MK2.5S machine
    Connect USB for MMU2S
    MMU2S is never recognized by Windows 10 PC
    Load Prusa firmware 3.8.0
    MMU2S immediately recognized by Windows 10 PC

**Expected behavior**
MMU2S unit should be a visible USB device to enable firmware flashing

**Screenshots**
n/a

**Serial Log**
n/a

**Additional context**
I was able to flash the MMU to TZB firmware while using Prusa MK2.5S firmware.  Following, I could flash the MK2.5S to TZB to enable further testing.  I also tried power cycling and resetting the printer.  Neither procedure allowed MMU to be recognized.  Every time I flashed back to Prusa MK2.5S version, MMU was immediately recognized.

TheZeroBeast, 2019-09-23:
When you attempt to flash the MMU2S hit the reset button on the printer just before clicking flash of the MMU2S FW in PRUSA Slicer. Let me know if that lets you flash the MMU2S.

habitforming11, 2019-09-23:
i successfully flashed it while prusa 3.80 was on the mk2.5. will that cause problems?

AbeFM, 2019-09-23:
Nope. It's about the reset. It's super helpful if you can access PC and printer at same time - hit "flash" and immediately (within a second or two) press the MMU's reset (not the printer's).

When reset, the MMU enters a programmable state for a few seconds, then leaves it and is programmable until the next reset.

TheZeroBeast, 2019-09-23:
Yes, I need to work on handling the comms mismatch without locking anything up.

TheZeroBeast, 2019-09-23:
I'll be doing up a Wiki flashing guide and a video of flashing too and from this FW, stay tuned.

TheZeroBeast, 2019-09-24:
Please regard the following tutorial regarding flashing issues.
https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/wiki/Flashing-Procedure

closed 2019-09-24 by TheZeroBeast