TheZeroBeast/TZB-MMU2S-Firmware #150: Support for Bondtech Mosquito

rmilyard, 2019-09-24:
I have a MK3S+MMU2S with Bondtech Mosquito extruder.  I have been having a lot of issues getting this all working together.  I was wondering if have tested this combo yet?

TheZeroBeast, 2019-09-24:
I only have the e3d v6 hotend combo sorry.

What issues are you having getting this going? I.E. what changes do you normally do to make it work on stock FW?

rmilyard, 2019-09-24:
My filament tips are stringy.  Also little to big.  They are getting jammed and not loading so have load fail until I cut the tips.

TheZeroBeast, 2019-09-24:
Thanks for the list of changes.

With the large tips and stringing, have you double checked you have the correct heatbreak and smaller inner diameter extruder PTFE? They are not optional.

I did a test a long time ago with a stock brand new e3d v6 heatbreak and not a single mmu2 load would work. I can't remember the tip diameter measurements but it was a substantial difference in tip size.

gizzburn, 2019-10-02:
Is the Bondtech BMG on the list to do next Robert?.

TheZeroBeast, 2019-10-02:
I'll compile and add the mosquito hex to my final release tonight.

rmilyard, 2019-10-03:
I would be interested also in new Bondtech Mosquito with Slice thermistor firmware.

TheZeroBeast, 2019-10-04:
Hi @rmilyard, [releases](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases) page now has a build with SE HT Thermistor tables and bondtech changes. Let me know if it works

TheZeroBeast, 2019-10-09:
Hi @rmilyard, any chance you've loaded the release for Mosquito to confirm it works?

rmilyard, 2019-10-09:
No. I am in CA with family at Disneyland.

TheZeroBeast, 2019-10-09:
Ah sweet! Enjoy, let me know when you return.

TheZeroBeast, 2019-10-25:
@rmilyard, how was Disneyland, any update on this thread?

TheZeroBeast, 2019-11-04:
Stale issue thread. It you have any issues please advise so we can re-open and resolve.

closed 2019-11-04 by TheZeroBeast