TheZeroBeast/TZB-MMU2S-Firmware #151: Go back to original

lmyvgta, 2019-09-25:
Hi ive installed the latest RC3 ver. I couldn't get to the original prusa firmware. Is there any updates that you can release so i can test both firmwares?

TheZeroBeast, 2019-09-25:
Hi @lmyvgta,

What do you mean you couldn't get to the original prusa firmware?

lmyvgta, 2019-09-25:
I couldn't reinstall the Original prusa firmware.  It didn't detect it.

TheZeroBeast, 2019-09-25:
Wonderful, you can try out the flashing wiki I made last night 👌👏

https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/wiki/Flashing-Procedure

Let me know how you go.

Did you test this firmware at all, or just wanted some stock loving?

lmyvgta, 2019-09-25:
Ill try, but ive just installed it and had some issues with the lenghts of the filament in the gears, and i didnt have time to set the new parameters. i thought it was easier to go back to stock, but i couldnt go back at all

TheZeroBeast, 2019-09-25:
Bowden calibration is the same procedure as stock 👍

Can’t get much easier than power and flash.

closed 2019-09-25 by TheZeroBeast

AbeFM, 2019-09-25:
> Ill try, but ive just installed it and had some issues with the lenghts of the filament in the gears, and i didnt have time to set the new parameters. i thought it was easier to go back to stock, but i couldnt go back at all

It's pretty trivial. Open door to see gears. Turn on printer, move selector to far right, press middle, left, middle, middle, left/right till adjusted right, middle, left.

That's it.

TheZeroBeast, 2019-09-25:
Indeed, here's the walkthrough I did up.
https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/wiki/Bowden-Length-Calibration