TheZeroBeast/TZB-MMU2S-Firmware #152: TZB-MK3S-Extruder Design

TheZeroBeast, 2019-09-27:


AbeFM, 2019-10-01:
Win! 
Link to Item: https://www.prusaprinters.org/prints/5756-tzb-mk3s-mmu2s-extruder-no-tower/

I had something to add but already forgot. Subscribed.

TheZeroBeast, 2019-10-06:
Let us collaborate on this design.
Propose/test/discus changes with my current design attached.

Apart from minimising the bondtech gear gap even further I don't have any issues with this design.
What do others think of the design?

[TZBMK3S-MK25S-Extruder.step.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3694329/TZBMK3S-MK25S-Extruder.step.zip)

CaiJonas, 2019-10-09:
Thank you for the files. I changed the lever part to make it little bit easier to print. Additioning to that I am using a metal pin. You will need a 3mm reamer for some holes. Let me know what you are thinking, I will do now a lot of testprints, especially with soft TPU Filaments. I am now using a titan heatbreak from RS Präzision(Germany). We will see how it will perform, until now I can say it looks very good. 

[TZBMK3S-MK25S-Extruder_CaiJonas.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3706773/TZBMK3S-MK25S-Extruder_CaiJonas.zip)

TheZeroBeast, 2019-10-13:
Hi @CaiJonas,
I've also been testing some changes I've done to reduce bondtech clearance as it was still over 1mm above gears which is unacceptably for flex.

Here is a photo of how little clearance I've gotten it down to....
[image]

So far I've been unable to jam TPU in it even when extruding fast and cool. CRAZY!!

I just reviewed your changes above and I like how you moved the lever to the other side so the tab prints flat on the bed, I'll incorporate in my next version.

I'd suggest modifying your design if you need to ream anything.
I still think the grubscrew and even use of a metal pin is over complicating the situation.
All well and good if you have it working although simpler the better for users printing and implementing the mod.

AbeFM, 2019-10-18:
I can't really view on the road, but in the past I've found overly
constricting the door's motion didn't help, maybe hurt, but not definitive.

My complaint is the gap between plastic and gears, especially on top,
sometimes unloads will wrap soft filament around the driven Bondtech.

MMU also suffers from unneeded big gaps.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

AbeFM, 2019-10-18:
Picture finally loaded! Looks like exactly what I was talking about. :-)

The other, very minor changes are to lever , the detailed shape of flag and
perhaps the angle of the surface the door pin pushes on.

Sorry if I'm not much help from out here. :-(

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

AbeFM, 2019-10-18:
I like what you did. Suggestions:
Small chamfer after gears (only on one side I think, not needed on lower
side)
Add meat above gears and let PTFE get closer.

Both are pretty small and might be hard to pull off.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

CaiJonas, 2019-10-19:
@TheZeroBeast we did a lot of research in our company with printing soft filaments. You are correct with reducing the gaps, additioning to that there are some other important things to do but it's not possible with the Prusa Mk3. We are able to print filaments with a hardness of shore a60. I have problems with loading the flex filaments. Sometimes it wraps around the gears in the main unit. Next step for me is changing the gaps in the main unit.

TheZeroBeast, 2019-10-20:
Yes, gaps in the MMU2S unit kill the flex at times, very large gaps around the gears.
I wonder why it was ever designed with such crazy tolerances. I've even seen bits of PLA jammed in around the gears in the MMU2S.

CaiJonas, 2019-10-20:
@TheZeroBeast Here is my design of the main unit. Didn't had the time to test it very much until now. You can change the tolerances of the insert, so it's not nessesary to print the hole unit again. Let me know what you are thinking.

Caution: You have to shorten the grub screws to 2mm

Best Regards
[mmu2-pulley-body_CaiJonas_9mm v2.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3748380/mmu2-pulley-body_CaiJonas_9mm.v2.zip)

habitforming11, 2019-10-26:
Do you have the proper length of the PTFE tubing that goes in the top of the hot end?  I'd like to make a cutting jig for it (if you don't expect it to change in the near future).

TheZeroBeast, 2019-10-26:
Hey @habitforming11,
I keep forgetting to measure when I have my extruder apart but will try get a length.

@CaiJonas, still getting to having a look at you step sorry.
What grub screws did you have to shorten to 2mm? The bondtech retaining grub screw?

AbeFM, 2019-11-12:
> @TheZeroBeast Here is my design of the main unit. Didn't had the time to test it very much until now. You can change the tolerances of the insert, so it's not nessesary to print the hole unit again. [...]

Why can't you do this all as one piece? Does the central section bolt in just to make it easier to test?

If you have access to square nuts, I'd suggest using them - I have the hex-nuts spin loose fairly often because I take out my aggression on small defenseless metric hardware. I'm certainly interested in trying it out if it at least works some. :-)

@TheZeroBeast What's the current state of the extruder? Is it "stable"? Does it use extra parts? Is the version on the Prusa site the one to use?

CaiJonas, 2019-11-12:
@AbeFM the central section is just for testing. Didn't had the time until now to test it very much. I'm fine with the m3 nuts, keep an eye on the suggested torque. A screwdriver beetween two fingers is more than enough :) If you want to do correctly use threated inserts for plastic. Feel free to test and optimise the design.

AbeFM, 2019-11-12:
Just a suggestion on the square nuts. I don't love the idea of adding in a less reliable feature that's there to increase reliability - my lived experience points to the square nuts working all the time and the hex's working most of the time, if you're careful. The heat-set stuff works, though then it's pick about print settings, you can't easily get them back, each one adds a chance of messing up the whole part, etc. Not bad for one in a part, troubling for 8.

The only difference in your outer-body are the nut retainers and the channel cut for the insert? Can you share the original CAD files?

TheZeroBeast, 2019-11-14:
> > CaiJonas : Here is my design of the main unit. Didn't had the time to test it very much until now. You can change the tolerances of the insert, so it's not nessesary to print the hole unit again. Let me know what you are thinking.
> > Caution: You have to shorten the grub screws to 2mm
> > Best Regards
> > [mmu2-pulley-body_CaiJonas_9mm v2.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3748380/mmu2-pulley-body_CaiJonas_9mm.v2.zip)
> 
> Why can't you do this all as one piece? Does the central section bolt in just to make it easier to test?

To test, as you thought. Although I wouldn't see a point to change it without some benefits to justify.

> 
> If you have access to square nuts, I'd suggest using them - I have the hex-nuts spin loose fairly often because I take out my aggression on small defenseless metric hardware. I'm certainly interested in trying it out if it at least works some. :-)

😂 too much torque, poor little guys. I think the hex is fine but be my gust if you need more grip 👍

> 
> @TheZeroBeast What's the current state of the extruder? Is it "stable"? Does it use extra parts? Is the version on the Prusa site the one to use?

The one on the Prusa site is the current stable release, extra parts, still a PIN and a flag.

CaiJonas, 2020-01-18:
Hi Robert,

Are there any news About your design? Look what I found on Aliexpress:

https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.9042647.0.0.287b4c4dHOVAWU&orderId=8009463021723304&productId=4000404170721

I’ve ordered one, Maybe it’s good. What do you think?

Best Regards
Cai

TheZeroBeast, 2020-02-18:
Hi @CaiJonas,

Busy start of the year, new Dad 👏👍

No progress on design apart from further testing on my now MK2.5S which is working flawlessly 🤘👍

TheZeroBeast, 2020-02-18:
Link update
https://www.prusaprinters.org/prints/7690-tzb-mk3smk25smmu2s-extruder-mmu2s-no-tower-v2

AbeFM, 2020-02-18:
A bit confused - you changed something? What? Is it just for the 2.5?

kajsilander78, 2020-02-18:
Could maybe make magnet-pockets a bit tighter. Works wonders so far! Have not looked (maybe done), turn hole for pin 180 degree in new placement. Add some meat around filament entrence to hide threads just for the look ;) Otherwise perfect! Suppose thats the reason this thread is so quiet....

AbeFM, 2020-02-19:
It's been working well for me. I feel there's a chance that the adjustment
screw should stand a bit more proud to ensure you're in the adjustment
range, the fan shroud is a little easy to monkey up, the pin is pretty hard
to get into place (bump could be smaller)... But the only real show stopper
is that the flag is marginal. It sounds like others haven't seen it but it
has been consistent between versions for me that the flag needs work. It
doesn't seem to cut the beam perpendicular to it, so it goes on-off-on. I
have yet to have one printed without mods that just worked.

The lever should have a bit more flat space to make sure it doesn't hit
pin. It would be nice to add a bump/feature so the wire holder off the back
doesn't rotate about its sole mounting screw.

Anyway, yeah the thing works and works well - it's a serious upgrade and it
is both what Prusa should have done and something they should change to.
Frankly, I'd hate to see them try, look what they did with my
filament buffer idea?!
Here's the lever I used. The flag is a hint too tall and is trim to fit -
you can see the cut away area on the pin side. I'm no STL editing hero, I
find them terrible to work with, but it gives you an idea - I tried to
attach it. :-)

sergiofagundes, 2020-04-26:
Hi, I'm using TZB firmware: 2.1.8-TZB77 commit 2386 with MM-control-01-2.1.8-RC2.

I have a Prusa MK3 bear full, not a S type. Can I still use the newer TZB firmwares on it or I need to upgrade my MK3 to MK3S and my MMU2 to MMU2S to use the newer firmwares. 

Thank you

sergiofagundes, 2020-04-29:
> Hi, I'm using TZB firmware: 2.1.8-TZB77 commit 2386 with MM-control-01-2.1.8-RC2.
> 
> I have a Prusa MK3 bear full, not a S type. Can I still use the newer TZB firmwares on it or I need to upgrade my MK3 to MK3S and my MMU2 to MMU2S to use the newer firmwares.
> 
> Thank you

Can I edit the configuration_prusa.h  and change the #define IR_SENSOR to #define PAT9125 and use the MK3S firmware on the MK3 and still use the MMU2S firmware ?

TheZeroBeast, 2020-04-29:
Sorry @sergiofagundes, been a busy few days. No the latest FW is not configured for the PAT91XX sensors anymore, a lot of code changed to support the IR_SENSOR which doesn't translate to the laser sensor.

sergiofagundes, 2020-04-30:
> Sorry @sergiofagundes, been a busy few days. No the latest FW is not configured for the PAT91XX sensors anymore, a lot of code changed to support the IR_SENSOR which doesn't translate to the laser sensor.

Hi, I'm now printing you new designed extruder for MK3S and will update the new firmware.

Is the manual bowden length calibration still necessary or it's like the original prusa mk3s mmu2s that calculates automatically the tube length ?

Thank you

sergiofagundes, 2020-05-11:
I have converted to the MKS3 using the TZB new extruder and new firmwares from TZB, but can't pass the calibration after the lcd shows to use a paper...  the firmware goes down and stops with a calibration failure. Some one can help with this ?

TheZeroBeast, 2020-05-11:
Open a thread and we can discuss @sergiofagundes 

Is the mmu Bowden length celebrated already?

sergiofagundes, 2020-12-20:
Any recommendations for a prusa bear full mk3s carriage and x axis compatible for this extruder on thingiverse ? Thanks

TheZeroBeast, 2020-12-21:
Hi @sergiofagundes,

I have not used any bear upgrades sorry mate. It is the same mount as stock PRUSA extruder so if the upgrade you find is compatible with stock the. You should be fine.

sergiofagundes, 2020-12-29:
Hi, can you share the source file of the extruder (tzbmk3s-back). I'd like to make a remix in Fusion 360 to use the BL Touch like this thing: https://www.thingiverse.com/thing:3561570

TheZeroBeast, 2021-01-22:
I'm working on modifying R6 Extruder(sneak peak) with my NO-TOWER and reduced filament path (yes R6 filament path is still not as tight as my last extruder design and I will be keeping this aspect for those that want less trouble with FLEX printing.
[image]

CaiJonas, 2021-01-23:
That looks good. Will you share the step file? I am using your old design with the dragon hotend. I am planning to use the dragonhotend without the upper part with the round mounting part. I want to bolt it direktly into the plastic.

Best Regards
Cai

Naugrimohtar, 2021-01-23:
I’m excited about this!

AbeFM, 2021-01-24:
Robert,
Is there anything else you're doing? Anything that would make the door
switch more sensitive and to more more freely would be great. I seem to be
able to adjust (through the use of set screw in the door) the set up to
work where I need it, but it's basically impossible for me to find a
setting that works for ninjaflex one day and PLA the next.

Maybe printing the body in CF play (for stiffness) or PETg for
slipperyness would help.... Anyway, curious for details about what you're
up to. :-)

TheZeroBeast, 2021-02-03:
@AbeFM, this is the nature of the method we're using to sense, for extremely soft filament the bondtech gears barely separate.
I reviewed the model and there isn't much room to change the pivot point to increase lever deflection.

TheZeroBeast, 2021-02-03:
I've finished installing the first version of TZB3-Extruder with BLTouch optional mount.
Testing tonight on my [TZB-MK3-FW](https://github.com/TheZeroBeast/TZB-MK3S-Firmware) that I have setup to use the BLTouch for MBL in conjunction with PINDA for initial calibration and XYZ homing.

I know, it's a bad photo but all I have currently, top plate needs revision asap as well.
[image]
[image]
[image]

AbeFM, 2021-02-03:
What do you expect to gain from the BL touch? I'd imagine it wouldn't work
too well on the textured bed - but it's giving me fantasies of "continue
print" by positioning probe over highest point and letting printer figure
out where to start the file from. :-)

TheZeroBeast, 2021-02-03:
It is supposed to be a lot more accurate and repeatable than the super pinda with std. deviation <10micron.

That's an interesting thought, find top of failed print haha

AbeFM, 2021-02-04:
In a recent firmware update a very cheap printer had a "continue from Z"
function - thought manually measuring is often a bit of a pain. It'd be a
bit of a task to shoehorn that in, but it's good to have rolling around in
the back of your head. :-)

I'm not against folks adding whatever they want.... But I've not had any
issues with z-height since the 7x7 grid thing. Just plain zero problems. I
do use the textured bed, perhaps that's part of it. Do you have issues?

Do you mind sharing if there are any other changes besides adding the
probe? I'd like to try it if there's some benefits, but if it's just adding
a probe I won't install I think I might skip it. :-)

sergiofagundes, 2021-02-04:
About Bltouch extruder, when will you release the stl files to print ? I want to use the Bltouch to use glass on my prusa lol... I know the flexible PEI sheets are nice, but when we are out of PEI sheets the glass is cheap and easily available anywhere. ;-)

TheZeroBeast, 2021-03-16:
Code and extruder design is in final stages of testing. I've been doing a bunch of TPU prints, still haven't jammed the extruder and the first layers are now spot on and temperature independent thanks to the BLTOUCH.

TheZeroBeast, 2021-03-16:
> In a recent firmware update a very cheap printer had a "continue from Z" function - thought manually measuring is often a bit of a pain. It'd be a bit of a task to shoehorn that in, but it's good to [...]

The 7x7 grid with PINDA2 is good when printing the same plastic with PINDA pre-heat gcodes but even when temp table is setup there is drift when going between low temp and high temp plastics.

Changes to add the probe are lengthy involving embedding the sensor into the xyz cal, z home and G28 meshbed leveling.
I chose to do it this way so that the xyz cal can still use PINDA to do skew/perp. correction.

AbeFM, 2021-03-16:
How do you deal with the powder coated beds?

I do actually HAVE a BL-touch. How many reads do you take, or, how long does getting the MBL done take now?

TheZeroBeast, 2021-03-16:
There is no dealing, it just works.

The probe count is set as you did for the PINDA but personally I use 3 probes.
I coded a high-sigma catch that checks that all probes (at a location) are within 25um and if not it starts that spot again, it can try 4 cycles on a location before an error is thrown.

That might be why the textured bed just works because the random times the probe does something funny due to the texture it is corrected/repeated.

I haven't timed it but it is comparable due to not needing to wait for PINDA temp, just bed then probe.

AbeFM, 2021-03-16:
Maybe the probe tip is fat compared to the texture. It seems crazy, but I
will go with "just works". Looking forward to trying it.

What do you print your extruders out of?

TheZeroBeast, 2021-03-16:
Yep, it's not very pointy.

PETG, that is still PrusaPETG from a roll I bought when I got the printers.

AbeFM, 2021-04-24:
Is this released yet? I broke my old extruder and would rather print the
new one. Where do I find it?

AbeFM, 2021-05-14:
> @AbeFM <https://github.com/AbeFM>, I've just released it here
> <https://www.prusaprinters.org/prints/66215-tzb-mk3smk25smmu2s-extruder-mmu2s-no-tower-v3>,
> if you would like to take a look.

Perhaps I will.

From the release notes:
V3 Changes

combine with R6 stock extruder, better airflow with the still superior flex
handling of the TZB extruder.
added an optional BLtouch mount (firmware tested about to be released with
support)

So it looks like it's not just the BLTouch mount. I thought the v2 also
used R6 but I guess not.

Also:
"Assemble as per the PRUSA manual for MK3S except for the magnets, they
need to both be installed backward to reverse the force of the lever."
Probably you only want to flip one magnet. That's what six years of Physics
in college does for you. :-)

I might bang one out, it'll be like the 4th thing printed I the new
extruder. :-) If you really wanted to excite me, put a hole for a set screw
so I can notch radiator/cold end and prevent it spinning. With my larger
fan I don't really have cooling issues anymore.
https://www.thingiverse.com/thing:3484443

As an aside: I'm printing non-hot parts in PLA so they flex less: Door,
Pin, and Lever. I think PET-g is to flexy. I may reprint those in cf pla
and am tempted to try the hot bits in ASA.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

gutschke, 2021-05-15:
I printed all the extruder parts in ASA. I love the super precise and smooth finish that I get with this material. And it has great heat resistance too. I printed the fan shroud from PC though, after managing to melt one that was made from ASA. Prusament PC Blend has even better heat resistance. I actually really like PC as well, but it doesn't have quite the same perfect finish that I get with ASA. It's close though.

I find that both materials print very easily on my MK3/S. In fact, other than the higher price, I think they are both superior to more traditional materials like PETG. You likely do need an enclosure though.

AbeFM, 2021-05-17:
I printed tinsy little things in ASA, and it works. I have a couple walls,
a foam panel, a top and a plexiglass front. It's not draft-free but it's
well minimized. I'm not sure if that's good enough.

You didn't do anything more than box, defaults on slicing for ASA and the
model as is out of the box? Even the little flag came out?

That would be reason enough to print a new one.

My ASA duct also eventually warped.

gutschke, 2021-05-17:
I have my printer dialed in really well, and I generally try to stick to filament sold by Prusa, as I know they have good profiles for their own filament. I print inside of the infamous IKEA enclosure, which helps with managing environmental temperatures and avoiding drafts. The upshot is that I don't really need to tweak anything and prints usually come out great without any adjustments.

I find that ASA prints just as easily as PLA, but it has a much better surface finish and tighter tolerances. I love working with this material. It's so nice. The PC Blend has a slightly worse surface finish and it handles just a little more difficult. I'd rate it as something between PLA and PETG in ease of handling; in other words, it's still really easy to print, but some parts don't come out perfect. On the plus side, it does have better heat resistance and I think a little better overall strength/longevity for heavily used functional prints. I use it for objects that get installed somewhere and then just need to keep working for a long time.

I don't quite recall what material I used for the flag. I think I did print it in ASA, but it's possible I just reused the old one that I had printed in PETG a while ago. That one did require a little bit of post-processing with sandpaper. I find that PETG is generally a little more annoying that way.

sergiofagundes, 2021-05-20:
> @AbeFM, I've just released it [here](https://www.prusaprinters.org/prints/66215-tzb-mk3smk25smmu2s-extruder-mmu2s-no-tower-v3), if you would like to take a look.

Can I use the old (tzb-mk3smk25smmu2s-extruder-mmu2s-no-tower-v2-stl) with only the 2 pieces for bltouch found in this new design or I need to print all the v3 new design ? Thank you

TheZeroBeast, 2021-05-20:
Hey mate, those V3 parts won’t fit V2 from memory.

gutschke, 2021-05-20:
Give it a try. It might work. I just did something similar. I already had the older design on my printer, but I wanted to upgraded to the newer fan shroud that clips into the fan holder. Just needed to print those two parts.

I didn't print the BLTouch version, but I think that would have been an option. You can always try and if it doesn't work then you print the remaining parts

sergiofagundes, 2021-05-25:
Hi Thanks. I have printed the 2 parts for bltouch, and putting the close the the old parts it looks like they will fit the old design. When I have a change I'll dismount and give a try. Thank you.

AbeFM, 2021-05-25:
> Hi Thanks. I have printed the 2 parts for bltouch, and putting them close

Two interesting bits:
1. Enclosing printer and printing ASA melted my tensioner, the spring
pulled the nut in leaving me no tension on bondtech. The flag also seemed
to bend, it always reads loaded now. Will check when I disassemble.

2. My bed sensor (pinda or finda or whatever) never fits in the clamp. I
put shrink tube around it, but would I be wrong in asking for that to be
smaller?

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes