TheZeroBeast/TZB-MMU2S-Firmware #153: TZB-MMU2S-Selector

TheZeroBeast, 2019-09-27:


XRyu, 2019-10-25:
Not sure if this is possibe, but what do you think about incorporating some modifications of this things into the TZB MMU2S Selector? :D
I really love the idea open FINDA Canal with (reused-stock (remix)) Magnets and the opened chamber for troubleshooting and understanding whats going on and even cleaning.
https://www.thingiverse.com/thing:3528314 (have a look on the remixes too)

TheZeroBeast, 2019-10-26:
Hi @XRyu,

Wow, that’s a bit different 🤔🙂
The only difference I really see is the open design to visualise the interaction.

It would be interesting but really imho my degrade part strength on that side resulting in a let functional part.

Which selector are you running at the moment if you don’t mind me asking

XRyu, 2019-10-26:
Hey @TheZeroBeast,

yeah, the open design is the key of this i think :)
I printed this one, but failed to assemble because i got the thing with the hole-to-cut wrong.
On the same day i found your project and printed the TZB-Selector (then v1) and use it since then.
Anyway, is it worth to print and update to the new one? :)

gizzburn, 2019-10-26:
I have printed this selector in ABS but have yet to fit it. might get time tomorrow to do it.

TheZeroBeast, 2019-10-27:
@XRyu, the new version of the TZB-Selector re-adds the blade holder and removes an unintended bit in the filament path so not worth redoing if you're still using the old one.

@gizzburn, just note if it doesn't go together that this is designed for PETG, for ABS you have to scale by 1.5-2% to have the same dimensional accuracy to the design.

gizzburn, 2019-10-27:
Robert I printed this on a PLA profile tweaked for ABS, I did not rescale it and it printed fine, I managed to fit to my MMU2s last night and did a test cube in pla, it came out perfect and the selector works a treat, you can here the ball cling to the magnet when the fillament retracts back to the buffer.
I printed off your buffer design and its working well now with my setup, I just need to read up on the Lin Advanced now as I see your new version now supports it, do I need to adjust anything for lin Advance once the firmware is updated.?

TheZeroBeast, 2021-01-22:
FYI peoples, I just uploaded a [new selector](https://www.thingiverse.com/thing:4732836). Not much different to last one I had but has some print improvements and blade support.

TheZeroBeast, 2021-11-29:
Yep, you’re spot on. I use a short cylinder magnet in mine. The dimensions would be on my Thingiverse post.

TheZeroBeast, 2022-02-06:
Hi @kajsilander78,

I have been experimenting with auto deplete with the end of spools and wow, I found out how the selector isn't setup to feed back a curly end of filament.

I have been prototyping a bunch over the weekend and think I might be onto something.
Hopefully it will improve allowing bent end filament both ways through the selector.

One design I tried would even get stuck attempting to load when filament wasn't straightened.
Impractical to say the least.

TheZeroBeast, 2022-02-06:
In the end I dropped the ball spherical cutout and made it a bit smaller as it was an 8.4mm sphere for a 7.9mm ball as well as making the main channel 2.3mm dia instead of 2.6mm.

The catch spot in either side of the ball hole. The idea of a low dropping ball means when loaded the ball is just touching the finda, being held in please.
When a curled end comes on either side it still lifts the ball (or that is the plan, testing today) and then as the ball doesn't expose the catch edge as it's being held from going too high by the finda it will pass by and into the following filament channel.

[image]

TheZeroBeast, 2022-02-13:
I few more design iterations and I have a reliable unload and load of end of spool curly filament.
Check it out [here](https://www.prusaprinters.org/prints/133891-mmu2s-tzb2-dual-blade-filament-path-refinement-mod) if you wish.

easonjim, 2022-10-25:
hi @TheZeroBeast, Can this selector open source design files? Since my blades are different sizes, I want to modify them for compatibility.