TheZeroBeast/TZB-MMU2S-Firmware #154: Drum loses home, re-homing incorrect until reset.

AbeFM, 2019-10-02:
**TZB-MMU2S Release Version**
2455/354

**Describe the bug**
On my mk3S/mmu2s if I say "home all" it consistently puts the selector on
pos A, the drum on pos B, feeds out the wrong filament next to selector.

If you stop it, it'll move on to the next one (I believe the printer
doesn't do any checks, so it just goes on), each time the drum is off by
one so you get all your filaments sticking out of the machine and the
selector moving over to snap them off.

I never liked how 'load all' worked, but it works worse now. :-P

**To Reproduce**
Also when loading, sometimes a failed load will lead to losing home.  It may be the drum is letting go after a failed load.

It will re-home and still be wrong. Repeated homings will be wrong each time.

I don't know if it's got too little current during the homing procedure or if it is getting lost.

Will update if I learn more, but wanted to get this out before I step out for a month.

TheZeroBeast, 2019-10-02:
Ok, well it'll be down to your detective skills as I don't have this issue.
I'd suggest loosen your tension on your drum maybe.

You could investigate the procedure in code and suggest an alternate approach.

AbeFM, 2019-10-02:
If I wanted to reproduce it, I would have failed loads. Unhook bowden tube. I didn't notice this with TPU. The drum tension is pretty low (2 turns past flush), which I've used for over a year and don't think it's the issue.

I can look at this in October.

TheZeroBeast, 2019-10-02:
Cool, do up some videos when you're back and we'll try find the culprit.

AbeFM, 2019-10-02:
It would help if someone other than me could disconnect that bowden and hit "load all" on the menu.

TheZeroBeast, 2019-10-02:
I'm curious why you're disconnecting your BOWDEN tube to do the load all?

TheZeroBeast, 2019-11-04:
Hey @AbeFM, can you try this out again and let me know?
I have been making some homing tuning that might make your idler homing more accurate

[TZB-MMU2S-3.0.3-ALPHA-BN363.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3802484/TZB-MMU2S-3.0.3-ALPHA-BN363.hex.zip)

AbeFM, 2019-11-04:
first thing tomorrow, sir!

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

AbeFM, 2019-11-04:
Trying this, along with the  TZB-MK3S-3.0.3-ALPHA-BN2476.hex.zip
<https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3804086/TZB-MK3S-3.0.3-ALPHA-BN2476.hex.zip>
posted
in "Rapid flashing LED on filament load (#163)", and while it's hardly
statistically significant, I'm 2 for 2 successfully loading compared to my
more recent track record of 1 out of 15.

I don't understand why it had suddenly got bad, probably a firmware change
I'd forgotten, but either way, it seems to be better now. I'll do an MMU
print when I get my spools untangled. :-)

Thanks again!

TheZeroBeast, 2019-11-05:
@AbeFM, cool, sounds good 👍

AbeFM, 2019-11-05:
5x5x3x2 = 150 changes in two prints with one failed unload that was my own fault is still a short test, but it says good things. So far I'd say it's better.

TheZeroBeast, 2019-11-06:
Looking good, should be solved by the sounds of it. 👌

TheZeroBeast, 2019-11-11:
Hope we're homing correct now, these last changes have been solid for me in the tropics.
I'll close this as it's resolved👍

closed 2019-11-11 by TheZeroBeast