TheZeroBeast/TZB-MMU2S-Firmware #155: Filament always ejected before start printing?

XRyu, 2019-10-05:
**TZB-MMU2S Release Version**
MK3S:MMU2S 3.0.0

**Describe the bug**
Question, is it normal this happens with TZB-Firmware?

Printing through Octoprint
-> Printer starts Bed Levelling, ejects filament and wants me change, asks if color correct
     ( i then mostly just cut a nice filament-end and put it back in)
-> Printer starts Bed Levelling again and then starts printing afterwards

Is this correct or is something not working correctly here?

TheZeroBeast, 2019-10-05:
This happens when the printer is thinking it’s running out of filament.

If you have your MMU2S FINDA set too for e it can be telling the printer it’s out of filament already.

Try turning it into the selector 1/4 of a turn more.

I wrote some notes on the FINDA in my [wiki](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/wiki/FINDA), it might assist.

Let me know if that helps.

XRyu, 2019-10-05:
Sorry, i think i was not clear enough :) This does happen when starting a print, not while a print is already happening. As of so, there won't be a filament activating the FINDA as it is not loaded to the nozzle at this point

TheZeroBeast, 2019-10-05:
Ok, I think I understand now.

So, do you have a channel loaded when starting the print?

MeshBedLevel(MBL) starts then at some point in stops
MMU2S ejects filament (M600 runout routine)
then MBL starts again from the start.

Could you also zip and share the gcode please, to double check where the T? command is being placed?

XRyu, 2019-10-06:
As far as i can see you nailed it, 

> #### Is this the sequence of events?
> 1. **MMU2S** Connected, Printer LCD displaying active channel, no channel loaded
> 2. Start Print (USB from OctoPrint)
> 3. Heaters are set and proceed to heat
> 4. MeshBedLevel(**MBL**) starts then at some point it stops
> 5. **MMU2S** ejects filament (**M600** runout routine)
> 6. **MBL** starts again from the start.
> 7. Print progresses fine
>    ....

This is exactly whats happening! :D I attached the gcode, maybe i oversee something :)
[Prusa_MMU_enclosure_PSU_holder_black_0.2mm_PET_MK3SMMU2S_2h46m.gcode.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3694381/Prusa_MMU_enclosure_PSU_holder_black_0.2mm_PET_MK3SMMU2S_2h46m.gcode.zip)

I have loaded all channels where there is filament, right now these were which i loaded: 1 PLA, 2 PETG, 5 PETG
I then set "Load filament 2" (not "to nozzle")

On Stock i was printing MMU okay, but had sometimes small problems with the IR-Sensor, but FINDA and PINDA always were fine.

TheZeroBeast, 2019-10-09:
Hi @XRyu,
Can you do a series of load_to_nozzle’s and ensure your IR_SENSOR is consistent in its readings?
I’m thinking if while mbl is happening it trips the IR_SENSOR then it’ll eject before continuing.

XRyu, 2019-10-09:
Hey, this might be it, i will try to check this later, as there is a print running right now.
Yesterday i did my first steps in Optimisation of my Start- and End-Gcodes in PrusaSlicer and inserted a phase for PINDA warmup and some settings Reset:

`; Original Prusa Start
M862.3 P "[printer_model]" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M115 U3.8.0 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode

; 2019-10-08TR Reset speed and extrusion rates
M200 D0 ; disable volumetric e
M220 S100 ; reset speed
M221 S{if layer_height >= 0.32}90{else}100{endif} ; compensate for thick layer heights

; 2019-10-08TR Set initial warmup temps
M104 S160 ; set extruder no-ooze temp
M140 S{max(first_layer_bed_temperature[0],65)}  ; set bed PINDA warmup temp
; Nozzle warmup before home to avoid driving hardened ooze into PEI surface
M109 S160 ; wait for extruder no-ooze warmup temp before mesh bed leveling, cool hot PINDA

; 2019-10-08TR Initial homing
G28 W ; home all without mesh bed level

; 2019-10-08TR PINDA warmup
G0 Z3; Raise nozzle before move
G0 X125 Y210 F10200; Move nozzle to PINDA warming position
G0 Z0.15 F10200; Lower nozzle to PINDA warming position
M860 S35 ; wait for PINDA temp to stabilize
G0 Z3; Raise nozzle before move

G80 ; mesh bed leveling
G81 ; check mesh leveling results
G0 Z5; Raise nozzle to avoid denting bed while nozzle heats
G21 ; set units to millimeters

M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
									   
; Send the filament type to the MMU2.0 unit.
; E stands for extruder number, F stands for filament type (0: default; 1:flex; 2: PVA)
M403 E0 F{"" + ((filament_type[0]=="FLEX") ? 1 : ((filament_type[0]=="PVA") ? 2 : 0))}
M403 E1 F{"" + ((filament_type[1]=="FLEX") ? 1 : ((filament_type[1]=="PVA") ? 2 : 0))}
M403 E2 F{"" + ((filament_type[2]=="FLEX") ? 1 : ((filament_type[2]=="PVA") ? 2 : 0))}
M403 E3 F{"" + ((filament_type[3]=="FLEX") ? 1 : ((filament_type[3]=="PVA") ? 2 : 0))}
M403 E4 F{"" + ((filament_type[4]=="FLEX") ? 1 : ((filament_type[4]=="PVA") ? 2 : 0))}

{if not has_single_extruder_multi_material_priming}
;go outside print area
G1 Y-3.0 F1000.0
G1 Z0.4 F1000.0
; select extruder
T[initial_tool]
; initial load
G1 X55.0 E29.0 F1073.0
G1 X5.0 E29.0 F1800.0
G1 X55.0 E8.0 F2000.0
G1 Z0.3 F1000.0
G92 E0.0
G1 X240.0 E25.0 F2200.0
G1 Y-2.0 F1000.0
G1 X55.0 E25 F1400.0
G1 Z0.20 F1000.0
G1 X5.0 E4.0 F1000.0
G92 E0.0
{endif}

M221 S{if layer_height<0.075}100{else}95{endif}
G92 E0.0
`

This is not much difference, but as i did this, i moved the MBL before the Filament-Loading
Its not much difference from Original-Prusa Settings other than this (left Original, right mine)

[image]

TheZeroBeast, 2019-10-09:
Yes, that'll totally do it. Nice work finding it. Happy printing.

closed 2019-10-09 by TheZeroBeast