TheZeroBeast/TZB-MMU2S-Firmware #156: MMU not loading correct filament

Penfold81, 2019-10-05:
Ensure you're running the latest **RELEASE** and you've checked the **Wiki** for answers before reporting. Any issues reported regarding previous **RELEASES** will automatically be closed and not responded to, this ensures the developer maintains focus & streamlines the development process.

**TZB-MMU2S Release Version**
TZB-MK3S-3.0.0-BN2460

**Describe the bug**
When starting a new print the mmu loads filament one regardless of what has been sliced in prusa slicer. It does a filament load after bed leveling, loads filament one then starts the prime line trying to push so much out that the extruder clicks

**To Reproduce**
Steps to reproduce the behavior:
Start a print using mmuv2
**Expected behavior**
A clear and concise description of what you expected to happen.

TheZeroBeast, 2019-10-05:
Hey @Penfold81,

Can you attach your gcode that was used so I can interrogate it?

Do you have the ability to provide a serial log of the start of this print to the point after it loses the incorrect filament number?

Penfold81, 2019-10-05:
Dave Ross has shared a OneDrive file with you. To view it, click the link below.
<https://1drv.ms/u/s!AoIQfQe1DY4CgSpkxbz1l1xFzA3z>
[https://r1.res.office365.com/owa/prem/images/dc-generic_20.png]<https://1drv.ms/u/s!AoIQfQe1DY4CgSpkxbz1l1xFzA3z>
snap_on_witches_hat_0.2mm_PLA_MK3SMMU2S_18h22m.gcode<https://1drv.ms/u/s!AoIQfQe1DY4CgSpkxbz1l1xFzA3z>

Here is the gcode

TheZeroBeast, 2019-10-05:
Hi Dave, thanks for the gcode so quick.

It all looks in order, can you describe the homing prior to the incorrect filament being loaded and what the lights do?

Or better yet a small video with your phone.

TheZeroBeast, 2019-10-20:
Hi @Penfold81, have you had a chance to look further at this issue you're having?

Penfold81, 2019-10-20:
Hi

I have tried a few things but decided to go back to stock as it’s a pain in the butt if using octoprint.

Thanks

Dave

TheZeroBeast, 2019-10-20:
Ah ok 👌
I use Octoprint for all my prints by the way, no pain in my butt.

closed 2019-10-20 by TheZeroBeast