TheZeroBeast/TZB-MMU2S-Firmware #157: OctoPi / OctoPrint is not opening anymore after TZB 3.0.0 firmware

chrisboer, 2019-10-06:
I updated my firmware towards TZB 3.0.0 but now I can not ready my RPi. 

I have put it on off / on again but still nothing happens.

I know it is pingable as I have assigned a dedicated IP for it.

So what is causing this issue.

chrisboer, 2019-10-06:
It finally connected after 15 minutes... must be the initializing which takes this long

closed 2019-10-06 by chrisboer

TheZeroBeast, 2019-10-06:
Hey @chrisboer,

Did you load the MK3S & MMU2S firmware’s before trying connect octoprint back to your printer?

What’s showing on the printers LCD?

TheZeroBeast, 2019-10-06:
Cool, good to hear.
Hope you enjoy the FW, let me know your thoughts and any improvement ideas 👍

chrisboer, 2019-10-06:
I will certainly test and help you... I am a little bit tired on Prusa firmware.

I now had a problem with loading filament, I did not change anything in the printer configuration uploaded your firmware restarted the printer and everything works ;-) So lots of issues are software related.