TheZeroBeast/TZB-MMU2S-Firmware #158: Printer not Restarting

gizzburn, 2019-10-09:
Ensure you're running the latest **RELEASE** and you've checked the **Wiki** for answers before reporting. Any issues reported regarding previous **RELEASES** will automatically be closed and not responded to, this ensures the developer maintains focus & streamlines the development process.

**TZB-MMU2S Release Version**
The version of the release relating to the bug report.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Serial Log**
Please attach and serial logs of this issue.

**Additional context**
Add any other context about the problem here.

Hi Robert, I have just tried to install the TZB-MK3S-3.0.0-BN2462-Bondtech-16.hex firmware to my printer, it installs ok but the printer never restarts, it just sits there with the onscreen message verifying 100%.
Tried it a few times but with the same results.
Printer is a MK3s + MMU2s with Bondtech BMG extruder.
Any ideas please..

TheZeroBeast, 2019-10-09:
Hey @gizzburn,
Did you get the MMU2S software on as well as until it’s on the printer side won’t boot.

gizzburn, 2019-10-09:
Ah no I did not Robert will try again and do both this time lol..

TheZeroBeast, 2019-10-09:
https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/wiki/Flashing-Procedure
Video I made on the flashing 👍

gizzburn, 2019-10-09:
I even watched that Robert as well, old age .

TheZeroBeast, 2019-10-09:
Yeah, I know that feeling 👍
Let me know if it all works with the bondtech bmg👍

gizzburn, 2019-10-09:
Will do

gizzburn, 2019-10-09:
Still no go Robert, the printer does not reboot after the update, I managed to get the MMU2s updated, but still the printer does not boot up fully, it justs sits there with 100% verifying

TheZeroBeast, 2019-10-09:
That is even after power cycling the printer I take it?
Interesting.... Let me have a think why that would be.

gizzburn, 2019-10-09:
Yes power it off and on makes no difference

TheZeroBeast, 2019-10-09:
[TZB-MK3S-3.0.1-RC1-BN2463-Bondtech-16.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3707351/TZB-MK3S-3.0.1-RC1-BN2463-Bondtech-16.hex.zip)

gizzburn, 2019-10-09:
Success Robert will run some tests now after No1 Calibration, thanks a lot...

gizzburn, 2019-10-09:
Robert just done a test cube,infact done two, one with the Bondtech firmware and one with your firmware,both cubes done on identical settings. I noticed on the Bondtech firmware the cube shows lines, I first noticed them on a lithopane I did, I printed the same one before the Bondtech firmware upgrade for a comparison.
But on your firmware those lines are much less visable, great work and thanks for your great Firmware.

closed 2019-10-09 by TheZeroBeast