TheZeroBeast/TZB-MMU2S-Firmware #159: MK2.5 bed unable to reach/maintain above 100C

habitforming11, 2019-10-09:
Ensure you're running the latest **RELEASE** and you've checked the **Wiki** for answers before reporting. Any issues reported regarding previous **RELEASES** will automatically be closed and not responded to, this ensures the developer maintains focus & streamlines the development process.

**TZB-MMU2S Release Version**
BN2462 / BN358

**Describe the bug**
Prior to using TSB firmware, printer had no problem reaching/holding 110C for ABS printing.  When using octoprint it wouldn't go above 101C.  Using the same gcode on SD card and it reached 106C.

**To Reproduce**
1. Initiate print with bed temp 105C first layer, 110C second layer.
2. After layer change, print bed will remain at 105-106C.

**Expected behavior**
Bed should reach 110C within a couple layers following layer 2.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Serial Log**
Please attach and serial logs of this issue.

**Additional context**
I also notice a clicking from either the power supply or heat bed, as the bed gets close to set temp (also noticed at 60C).  This is much more pronounced than I ever heard on Prusa 3.7.1 or 3.8.0 firmware, to my memory.  I will reflash prusa firmware to confirm both the clicking, and if the overall temp issue persists.

TheZeroBeast, 2019-10-10:
Morning @habitforming11,

🤦‍♂️😂 I saw that in the MK2.5S config yesterday and it crossed my mind I might need to adjust as I did with the MK3S.

I'm checking HeatBedPIDs now, I'll update you in a bit.

The clicking is normal on my FW as I approach PWM a different way to PRUSA.

habitforming11, 2019-10-10:
Cheers!  I was just going to update that by changing back to stock FW the heating issue went away, as well as the clicking.

Seperately, could my loading issues with your firmware been contributed to the longer internal path of the Bear extuder, vs the newer Prusa design (50mm vs 43mm, I believe)?  I use a custom bear firmware that has those changes in the stock firmware.  I'm rather embarrassed this didn't occur to me earlier.

TheZeroBeast, 2019-10-10:
Here we go,

I should of aligned the BEDPIDs with the MK3S ones due to my PWM timer changes.

[TZB-MK25S-3.0.1-BN2466.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3710160/TZB-MK25S-3.0.1-BN2466.hex.zip)

[Here's](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/issues/130) where I had my scope on the bed setting it up.

Yeah mate, that'd make a big difference, what bear changes do you make, I'll add it to Precompiled binaries.

TheZeroBeast, 2019-10-10:
BN2466 Pushed to release page for other MK2.5S owners.

habitforming11, 2019-10-10:
Thanks, I'll confirm in about 1.5 hours.  Have a print going now.

habitforming11, 2019-10-10:
The firmware changes for the Bear extruder are on the following link, under "Changes from Original Prusa firmware"

https://github.com/bear-lab-3d/Prusa-Firmware/releases/

TheZeroBeast, 2019-10-10:
Bear issue now in [release](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/releases/tag/3.0.1) page, let me know if it helps although it only changes some Z axis offsets.
If the extruder design is different, as in different distance from bondtech gears to nozzle then adjustments would need to be made for it to be smooth.

The heating issue should be all good now though👍

habitforming11, 2019-10-10:
I just tested out the new Bear firmware TZB-MK25S-3.0.1-BN2467-Bear.hex, and the heat bed issue is still present.  won't go past 102C, and still makes a clicking noise.  is there a bed PID calibration I need to run?

TheZeroBeast, 2019-10-10:
What set bedtemp are you trying to obtain?
There is a autotune you can run but I don't think that'll help with the way PID is calculated in PRUSA FW, I'll review options based on the first question.

habitforming11, 2019-10-10:
Trying to get to 105 to start the print, and 110 for layer 2+.

TheZeroBeast, 2019-10-10:
Hey @habitforming11, found the issue for sure this time.
Interesting that I haven't had the issue as yet but I'll put that down to my enclosure when printing at higher bed temperatures.

Attached for your consideration 😀
[TZB-MK25S-3.0.1-BN2468-Bear.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3713800/TZB-MK25S-3.0.1-BN2468-Bear.hex.zip)

TheZeroBeast, 2019-10-13:
Hi @habitforming11, did you have a chance to test?
I did a heated bed run on my mk2.5 with the fix and it’s heating fine to 110c👌

habitforming11, 2019-10-13:
hey bud I'm going to try to get to it this evening.  had some prints i had to get done over the past few days

habitforming11, 2019-10-14:
I'm still getting the buzzing sound from the main board / heat bed.  Video attached below.
https://drive.google.com/file/d/1DD9WKI_0YhG6Upx-w8VGNE4n-8KUIHQo/view?usp=sharing

The bed does reach 115 without issue, and with zero overshoot.

TheZeroBeast, 2019-10-14:
Yes, the buzz is normal due to hardware limitations of board design requiring low frequency switching 30Hz, PRUSA do a lot of hacking to make it not buzz which uses a lot of CPU cycles, much cleaner to accept the buzz. Awesome to hear it now reaches temp 👍

closed 2019-10-14 by TheZeroBeast

habitforming11, 2019-10-14:
Ok I can live with the buzz, just want to be sure it's expected and doesn't cause undue wear.  Thanks again for all of your effort with getting this going.  Testing to resume in a few days :)