TheZeroBeast/TZB-MMU2S-Firmware #160: Show PINDA-temp when heating

XRyu, 2019-10-23:
Is it possible to show in the LCD when PINDA is heating?

I have a pre-print gcode sequence for PINDA-warmup before Mesh Leveling,
when this happens the Terminal shows the Temperature of PINDA, but the LCD only shows "Please Wait", could this be enriched by PINDA-Temp?

I use this GCode to wait for PINDA-Temp to stabilize:
M860 S35 ; wait for PINDA temp to stabilize

TheZeroBeast, 2019-10-23:
Yes 👍

TheZeroBeast, 2019-10-27:
Hi @XRyu, just an update, I'm testing this feature at the moment.
I made the status line of the lcd show PINDA current/target temps until M860 is complete.

TheZeroBeast, 2019-10-27:
Try this out when you get a minute @XRyu 
I haven't found a fix for the lcd flicker every 1s update but I'll keep an eye out. Still nice and functional.

[TZB-MK3S-3.0.3-ALPHA-BN2473.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3775551/TZB-MK3S-3.0.3-ALPHA-BN2473.hex.zip)

XRyu, 2019-10-29:
Works and looks pretty nice @TheZeroBeast  :D
I see what you mean by the flickering, it could be smoother, but its fine as long its needed :D

TheZeroBeast, 2019-11-01:
Ok, so it worked but not when printing from SDCard so I tried again and I think this solution is much better.

[TZB-MK3S-3.0.3-ALPHA-BN2474.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3796923/TZB-MK3S-3.0.3-ALPHA-BN2474.hex.zip)

XRyu, 2019-11-02:
Great! I will try this later today :)

XRyu, 2019-11-03:
Hah nice one! :D First i thought "heh? its just Please Wait again..." and then i've seen it - beautifully integrated :D

TheZeroBeast, 2019-11-04:
😂 awesome, i'll close this off but it'll be in the 3.0.3 release when it comes 👍

closed 2019-11-04 by TheZeroBeast

AbeFM, 2019-11-05:
Out of curiosity, why do you need it? I used to have issues with preheating
but with 7x7 things just stick.

What's your experience been?

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

XRyu, 2019-11-05:
Hey @AbeFM , i have a PINDA Warmup in my Start-Gcodes. Everything worked and works, i just thought "there is this process i know is going on, but printer only says 'please wait', it knows all the values, so it should communicate the current progress to me" :)
so this was more a feature / cosmetic request

AbeFM, 2019-11-05:
That's what it looked like, and I also think it is a reasonable request.

But I was saying, if you just forget the warm up, how does it work? I used
to wait a LONG time for preheat, but once the 7x7 MBL was added to the
official PR firmware I just stopped waiting for temperature and I had great
first layers.

If you turn off the wait/check, what happens?

XRyu, 2019-11-05:
Hmm i dont know didn't try. I've set to 7x7 two days after i got my prusa, so i might not have much of an reference. But my first layers got better when my PINDA is warmed up, before i had some errors mainly in the left side of my bed. Which feels like it takes only  2-3 minutes, which is finally pretty fine for me for any print :)
( i have open to-do to try the Bed-Leveling Method with Nylocs to get it even better )

AbeFM, 2019-11-05:
On a related note - I think that's way overkill. YMMV, but I loosened the
front and rear y-axis mounting screws, and lifted the bed a little where it
was low, dropped a little where it was high and got the corrections needed
down to VERY small numbers.

You may find this works just as easily for you. When I mention it online, I
get a lot of people telling me that just because their hours long process
worked, mine can't... But it does. :-)

I'm not against heating up the sensor, but I am against even 2 more minutes
on every print just to make up for improper alignment.

XRyu, 2019-11-05:
i believe your prcoess can work :) as of right now i am pretty happy with my printer and print quality, but you know, one always wants to upgrade or make better :D
 i dont really know how bad my bed i levelled though, need much more reading before changing anything. currently it looks like this in my Octoprint
[image]