TheZeroBeast/TZB-MMU2S-Firmware #161: Marlin 2.0 Support

gorbi87, 2019-10-26:
Hi,
i have selfbuild CoreXY with Marlin2.0 and MMU.I try to change the mmu firmware from prusa to TZB. Flashing sucessful but no serial connection after flashing between mmu and einsy. Unfortunately i cant use the Prusa firmware because CoreXY is not really supportet. With the original Prusa MMU firmware everthing works fine. Any chance to support Marlin2.0 too?

TheZeroBeast, 2019-10-27:
Hi @gorbi87,

I don't have the hardware or time to support marlin2.0 sorry.
It would be a large amount of work to pull the two together, not impossible though.

What approach did you take?

gorbi87, 2019-10-27:
Thanks for your answer. I have exactly the same hardware like prusa but in a CoreXY build. Everything works fine with Marlin 2.0 and the MMU2s with the original Prusa firmware. But when i use your firmware for the mmu it seems there is no serial connection between the Einsy board and the mmu. So i thought you have change something in the serial communication in the mmu firmware. Maybe you have any Idea what can i try and maybe recompile it. Thank you

TheZeroBeast, 2019-10-28:
Sorry, this firmware doesn’t communicate the same as PRUSA FW if you’ve only flashed the mmu2s.
Marlin would have to be retrofitted with my uart setup include payload protocol as well as all the code associated that understands the mmu2s messages.

TheZeroBeast, 2019-10-28:
It would make it easier having the same hardware.
The entire mmu2s and comms setup is different to stock

On marlin side you’d start by pulling in the mmu.cpp, mmu.h, uart2.cpp & uart2.h files.
These have the serial setup, the protocol methods and most of the mmu interface code.

gorbi87, 2019-10-29:
ok, thank you. I will try this

TheZeroBeast, 2019-11-04:
I'll close this issue thread but feel free to post back any questions, I'll reopen and try assist.

closed 2019-11-04 by TheZeroBeast