TheZeroBeast/TZB-MMU2S-Firmware #162: MMU2s Selector making loud click on startup

gizzburn, 2019-10-28:
Just updated to latest Bondtech BMG software TZB-MK3S-3.0.2-BN2472-Bondtech-16.hex and TZB-MMU2S-3.0.2-BN362.hex.

Now everytime the printer starts and the mmu2 boots up it sounds like its jamming, its the motor that moves the roller with the bearings on, its like its want to move the roller but has reached the stop and makes this horrible noise, was fine till the update running on the previous version.

Any ideas Robert as it wont change the filament now as well to the next colour.

TheZeroBeast, 2019-10-28:
I’ll double check I haven’t changed something that matters for it 😅👍

gizzburn, 2019-10-28:
Also Robert are the ramming and cooling settings still the same for the my setup? below are my settings
[image]

TheZeroBeast, 2019-10-28:
Did you upgrade from v3.0.1 @gizzburn?

The normal stock speeds and distances work on a normal setup, for the different extruder I’d wager you’d need to adjust them, not that I’ve seen what you’d have to adjust for the bmg

gizzburn, 2019-10-29:
Yes Robert I upgraded from 3.0.1.

TheZeroBeast, 2019-10-29:
Cool, I found the culprit, I’ll send through a test hex soon 👍

TheZeroBeast, 2019-10-29:
Give this a whirl @gizzburn, let me know if it's still not homing properly for the idler 
[TZB-MMU2S-3.0.3-ALPHA-BN363.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3783187/TZB-MMU2S-3.0.3-ALPHA-BN363.hex.zip)

gizzburn, 2019-10-29:
Robert it will not flash the file, tried several times, even tried both files printer and MMU but it wont flash the file. I tried the old version file for the MMU2 and that flashed no problem, The log says time out on port 8 and it uses the same port for the old file.???

Panayiotis-git, 2019-10-29:
Try adding the line:
; device = mm-control
at the beginning of the hex file.

TheZeroBeast, 2019-10-30:
Yeah, that line was in the hex, I just double checked. See if it shows for you on this one, same zip I downloaded from the above link.

[TZB-MMU2S-3.0.3-ALPHA-BN363-Retry.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3786333/TZB-MMU2S-3.0.3-ALPHA-BN363-Retry.hex.zip)

gizzburn, 2019-10-30:
Ok Robert that worked and I now get three taps and no grinding noise, will try a print.

TheZeroBeast, 2019-11-01:
Hi @gizzburn, I did some further tuning, also going back to two touches not so fast as the fast ones weren't accurate. Can you please let me know if this makes the load clicking on startup?

[TZB-MMU2S-3.0.3-ALPHA-BN363.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3796934/TZB-MMU2S-3.0.3-ALPHA-BN363.hex.zip)

TheZeroBeast, 2019-11-06:
@gizzburn, have you had a chance to try the last hex I posted above?

gizzburn, 2019-11-06:
Hi Robert will try it today been away for a few days..., but before I left I could not get the printer to print at all, on multi colour, it would not unload the filament at first so the first filament got stuck.
I did some checking and I have found on the bondtech extruder body thats supplied with the upgrade as a very small bore hole from the ptfe on top throught to the ptfe in the heatbreak. I cured this by opening it up a little.

But still get problems now with the filament loading, it gets to the IR sensor and starts to print but nothing comes out, I have tried a few things like recall the sensors and the bowden tube length but no luck as yet, but will keep trying to find the problem.

Will let you know when I have installed the latest above.

gizzburn, 2019-11-06:
Robert I think there might be a bug, I cant print anything because my temps are going up and down even after a PID tune, it drops to 204 from 230 at random times, I am going to try Bondtechs FW and see if that does the same.

TheZeroBeast, 2019-11-07:
Hi @gizzburn,

If you IR_SENSOR is trowing a false positive it's going to break the load sequence.
I upgraded my extruder to my latest design a couple weeks ago and needed to retune my sensor. It was trowing false positive once every rotate because the grub screw, I had to file the corners away a little bit more. Just an example how fine the adjustment is to get solid sensor readings.

> Hi Robert will try it today been away for a few days..., but before I left I could not get the printer to print at all, on multi colour, it would not unload the filament at first so the first filament got stuck.
> I did some checking and I have found on the bondtech extruder body thats supplied with the upgrade has a very small bore hole from the ptfe on top through to the ptfe in the heatbreak. I cured this by opening it up a little.
> 
> But still get problems now with the filament loading, it gets to the IR sensor and starts to print but nothing comes out, I have tried a few things like recall the sensors and the bowden tube length but no luck as yet, but will keep trying to find the problem.
> 
> Will let you know when I have installed the latest above.

Your extruder having erratic temp readings like that screams broken wire on your heater of thermistor, you can do a simple check of this by preheating it and then moving it left and right by hand, see if the reading varies, then do Z up and down from printer menu or serial connection.

gizzburn, 2019-11-07:
Robert it was my install of slicr3 making the temps alter, I used an older version and ithe temps were good, I also had a partial clog in the heatbreak. I have stripped it down and cleaned it out, all built up now.

I also keep getting this on the lcd and the printer reboots.
[image]

I will install your latest FW tomorrow and do some more tests hopefully it should be ok now.

gizzburn, 2019-11-09:
Robert I kept getting this screen even with the new version and random reboots, so I have installed the Bondtech FW the latest from then and not seen this screen or the random reboots.. But still I cant print multi filaments. 
I noticed when watching the sensors that when it goes to make the filament change the IR sensor is working as it should, but the finda never changes from 1 on filament change, but does go to 0 when I stop the print and it parks the filament. Should it not go to 0 then back to 1 when changing the filament .?
 I was wondering if I should try a factory reset on the MMU2 as I have already done it on the Mk3s.?.

gizzburn, 2019-11-09:
Ok did a factory reset on the MMU2 and installed the latest TZb FW up to now no resets and no screen with jumbled up text.
If you remember Robert I printed off an updated finda holder with a magnet to pull the ball bearing down on filament unload, I thought this could be my loading problem, so I went back to the original holder but this made no difference.
I am at a loss now as I have tried all I can think of to get it to print more than one colour. It prints a single colour fine, but when it has to load multiple times it will not work.
Most of the time it fails when the filament reaches the bondtech gears, the led light on my IR sensor lights up but it does not extrude any filament, if slacken the spring door and  I pull the filament out and then insert it manually it will then pull the filament in and print, then it might load the next ok, but fail on the next one.

I am not sure about the finda probe as it stays a solid 1 when changing filament, I have tried to search for an answer if this correct but did not find anything.

Just printed off a test cube and its perfect..

gizzburn, 2019-11-09:
Robert just been reading this page of yours https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/wiki/FINDA.

When I checked my Finda it works the opposite way to the way you say how to check it. When I put the filament in and press the finda down on the ball and filament my light is off, when I pull it away the light comes on, which way is correct, I would say your way but how do I change it?.

TheZeroBeast, 2019-11-09:
Hey @gizzburn,

Glad to hear a factory reset fixed the screen thing, I hadn’t seen that before.

The finda response is very interesting, I’ll double check my finda light response soon and get back to you.

gizzburn, 2019-11-09:
Robert found this post, on githup it seems there is a problem on Finda probe as this is eaxctly what I am seeing on my printer.

https://github.com/prusa3d/Prusa-Firmware/issues/1770

TheZeroBeast, 2019-11-11:
Hey @gizzburn,

I double checked and you're right 🤦‍♂️😂 the LED is opposite to our PINDAs.

Regarding that prusa issue #1770 I don't feel it's related, I run an entirely different comms protocol between printer and MMU2S.

Unless your MMU2S is loosing comms mid print (comms cable fault) then it'll be updating.
If you still have comms then your MMU2S will be responding to commands/gcodes.

If the issue persists I can enable debug on FINDA for you in a custom build for troubleshooting the issue.

As this #162 is now resolved I'll close it off. Next time you have the FINDA issue note all you can and open a new issue so we can look further 👍

closed 2019-11-11 by TheZeroBeast