TheZeroBeast/TZB-MMU2S-Firmware #163: Rapid flashing LED on filament cold load

Canceltodebug, 2019-11-01:
Ensure you're running the latest **RELEASE** and you've checked the **Wiki** for answers before reporting. Any issues reported regarding previous **RELEASES** will automatically be closed and not responded to, this ensures the developer maintains focus & streamlines the development process.

**TZB-MMU2S Release Version**
3.0.2

**Describe the bug**
Rapid LED (first light) flashing when filament loads through to extruder when cold. Bed and extruder continue to reach temp and no error message.

**To Reproduce**
Load filament from MMU2 to extruder when cold.

**Expected behavior**
No flashing LED and print to start

TheZeroBeast, 2019-11-04:
Hi @Canceltodebug,

No new features regarding this have been implemented since 3.0.0.
The flashing LED is indicating load error which needs to be cleared before it will continue.

How is it even attempting to load if the hotend isn't up to temp? This is indicating that your start gcode isn't correct.

If by chance you're using the octoprint script that replaces T? with T<n> then this will cause this issue as T<n> requires a preheated hotend, to fix adjust the position of T? in your slicers start gcode to after hotend is heated.

Canceltodebug, 2019-11-04:
I've rolled back to 3.0.1 and it is working as expected again.
No modifications are being made to the gcode after being sliced with PrusaSlicer 2.1.0.
Not using Octoprint either. Just a standard 'model > slice > print from SD card'.

I will re-install the 3.0.2 firmware again later today but since I'm not changing anything during the slicing process from 3.0.1 to 3.0.2, I don't see if this will help as it sounds like the gcode is being misread somehow? I'm not making any modifications as all.

TheZeroBeast, 2019-11-04:
Hi @Canceltodebug, I think I found a line that might be causing this issue that crept in with LA1.5 integration. I'll build a test hex tomorrow for you to try.

Canceltodebug, 2019-11-04:
That's great, thank you.

TheZeroBeast, 2019-11-04:
[TZB-MK3S-3.0.3-ALPHA-BN2476.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3804086/TZB-MK3S-3.0.3-ALPHA-BN2476.hex.zip)

AbeFM, 2019-11-04:
does this certain include the changes you wanted me to test yesterday?
Should I run this one instead of what's linked in other thread?

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

Canceltodebug, 2019-11-04:
> 
> 
> [TZB-MK3S-3.0.3-ALPHA-BN2476.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3804086/TZB-MK3S-3.0.3-ALPHA-BN2476.hex.zip)

Loading as per 3.0.1 again. Thank you :)

TheZeroBeast, 2019-11-04:
@AbeFM this one mate.

@Canceltodebug, did you see if that build helped the lid issue?

Canceltodebug, 2019-11-05:
Hi @TheZeroBeast 
Yes, it is loading correctly again. Thank you.

TheZeroBeast, 2019-11-06:
Wonderful, the fix will be apart of the next release, thank you for your help.

closed 2019-11-06 by TheZeroBeast