TheZeroBeast/TZB-MMU2S-Firmware #164: OLED Screen Support in Firmware

alackofcolor, 2019-11-04:
Any chance of adding support for WEH002004 OLED display, like in gussner's firmware? Thanks.

TheZeroBeast, 2019-11-04:
Do you have a reference I can look at of Gussner's?

What nifty benifits does this OLED provide, I might also be interested.

3d-gussner, 2019-11-04:
@TheZeroBeast Hi, here the reference code and some other links
https://github.com/3d-gussner/Prusa-Firmware/tree/MK3_WEH002004_OLED
https://forum.prusaprinters.org/forum/original-prusa-i3-mk2-s-others-archive/oled-replacement/#post-31385
https://vmod.wordpress.com/%E2%80%A6/%E2%80%A6/mk2s-oled-screen-replacement/

https://www.facebook.com/groups/284271408664100/search/?query=oled&epa=SEARCH_BOX

alackofcolor, 2019-11-04:
@TheZeroBeast  @3d-gussner If I'm not mistaken, most of the changes in the firmware need to be done to the lcd and ultralcd files. Versions of those files for OLED screens had been floating around on the forums for a while, until Prusa FW 3.8 or so, which didn't play nice with them anymore. That's when I came across the Zaribo and Gussner FW, which added support for those screens in a real firmware release.

As far as the benefits, they look much nicer with higher contrast and feel much more responsive to me. There are quite a few people using them now as it's a pretty cheap upgrade.

3d-gussner, 2019-11-04:
@alackofcolor Pretty cheap upgrade is relative as it costs ~50USD https://zaribo.com/home/245-oled-display-20x4-white-black-by-ldo.html :-)

3d-gussner, 2019-11-04:
BTW: Some Zaribo users and the owner are testing the OLED firmware also on LCD displays. I don't own one and accidentally flashed my MK3s with the OLED firmware and it worked. We will see if that also works in the "wild" :-)

alackofcolor, 2019-11-04:
@3d-gussner I believe I got mine for $35 off Digikey. It's such a common electronic component, that electronic sellers won't inflate the prices like 3D printing sellers will.

TheZeroBeast, 2019-11-04:
Looks cool, quite steep price point for a small OLED

I'll have to do some review and testing of the code. Thanks @3d-gussner for linking in all this info

TheZeroBeast, 2019-11-27:
@alackofcolor, as per your other thread I've been testing and will post hex with OLED code encorporated soon.

TheZeroBeast, 2020-02-18:
Testing has been solid with this, I'm in the process of pulling us up to Stock-3.9.0-RC1, I will include this enhancement in that.

TheZeroBeast, 2020-03-09:
This is now implemented on latest release with 3.9.0-RC1 upstream rebase.

closed 2020-03-09 by TheZeroBeast