TheZeroBeast/TZB-MMU2S-Firmware #165: MK3S Bondtech Mosquito Support

alackofcolor, 2019-11-04:
Request to add MK3S Bondtech + Mosquito printer support.

TheZeroBeast, 2019-11-04:
Hi @alackofcolor,

as per issue #150 this should already be implemented under released compiled binaries, the one ending in **Bondtech-16-SE_HT_Thermistor**

If you could let me know if you have any issues, it has all the changes I understand is required for the mosquito

alackofcolor, 2019-11-04:
@TheZeroBeast 

According the the [Bondtech website](https://www.bondtech.se/en/2019/09/12/firmware-updates-for-prusa/), they release firmware modifications for the Bondtech + Mosquito combo, as well as Bondtech + Mosquito + HT Thermistor.

For people simply swapping out the E3D v6 for the Mosquito, but sticking with the E3D thermistor that came with the printer since it's already Mosquito-compatible, the Bondtech-16-SE_HT_Thermistor version will not apply.

TheZeroBeast, 2019-11-04:
Ah ok, well the **Bondtech-16** is the version without SE HT Thermistor table enabled.

alackofcolor, 2019-11-04:
Does the Bondtech-16 version account for the Mosquito as well, or just the Bondtech extruder? It looks like the Bondtech website's firmware shows changes that are specific to the Mosquito (path lengths, etc), even when the normal, non-HT thermistor is used. Thanks.

TheZeroBeast, 2019-11-06:
So many 😅 I had no idea how many bondtech variants were out there and yes the one I currently do appears to be for the BMG with the stock V6 hotend.

I've pulled in the mosquito variants and attached a binary build for MK3S mosquito for you to test.

[TZB-MK3S-3.0.3-ALPHA-BN2477-Bondtech-16-Mosquito.hex.zip](https://github.com/TheZeroBeast/TZB-MMU2S-Firmware/files/3812278/TZB-MK3S-3.0.3-ALPHA-BN2477-Bondtech-16-Mosquito.hex.zip)

alackofcolor, 2019-11-06:
Yeah, there's quite a few out there. Luckily it seems like Bondtech does a pretty good job of keeping each of the firmware variants updated. Though I'm not sure what it involves on your end to tie that into what you're doing.

I'll be able to test the firmware you attached in the coming days and get back to you. Thanks for the work.

TheZeroBeast, 2019-11-16:
Any chance to test this mate?

alackofcolor, 2019-11-17:
Unfortunately not yet. It took a while to upgrade my printer with the Bondtech+Mosquito because the thermal paste shipped separately and arrived a while later. I finally got the upgrade installed and wanted to make sure everything was working first on stock-ish firmware. So I'm running Gussner's OLED firmware, with Bondtech's changes for the new extruder added in.

One thing I'm concerned about in testing the firmware you previously attached is that it's not for the OLED screen, and non-OLED firmwares eventually scramble the screen text and make the display impossible to navigate. But I might be able to get some prints started with your firmware and see how they turn out without having to use the display. Another alternative is you supplying the source for the compiled hex you attached previously, and I can see if I can make the OLED changes to your firmware. From my understanding, the OLED changes only affect a couple files, so it shouldn't be hard to do, and it shouldn't affect the other aspects of your firmware.

3d-gussner, 2019-11-17:
@alackofcolor and @TheZeroBeast We had to make a new firmware for new Bondtech Extruders as they are bit longer than the first one. Some users had calibration issues. Checkout:
- For OLED firmware which also seam to work with LCD screen just fine
  - https://github.com/prusa3d/Prusa-Firmware/compare/v3.8.1...3d-gussner:MK3_WEH002004_OLED
- For Bondtech firmware changes especially MMU
  - https://github.com/Zaribo/Zaribo-Firmware/tree/Bondtech_Prusa-Firmware
  - https://github.com/Zaribo/Zaribo-Firmware/tree/Zaribo-MK3
  - https://github.com/BondtechAB/Bondtech_Prusa_i3/tree/master/Firmware/3.8.1
  -https://github.com/BondtechAB/Bondtech_Prusa_i3/commit/37b7033617dde62531d7438f67b56f5c39bfb9bb
  - https://github.com/prusa3d/Prusa-Firmware/compare/v3.8.1...Zaribo:Bondtech_Prusa-Firmware

Please also checkout
-hhttps://github.com/Zaribo/Zaribo-Firmware/blob/Zaribo-MK3/Firmware/Marlin_main.cpp#L2135
as the Bondtech Extruder is based on Prusa Extruder R3 which is bit longer than Prusa Extruder R4

@TheZeroBeast vertigo235 created a nice script which I modified to the needs of Zaribo firmware:
- https://github.com/Zaribo/Zaribo-Firmware/blob/Zaribo-MK3/Firmware/variants/create-variants.sh
Helps a lot to generate variants from slightly modified "stock" variants. And compile firmware variants using `PF-build.sh` bash script.

TheZeroBeast, 2019-11-27:
Hi @alackofcolor, I have been testing the OLED updates with my codebase over the past week and all appears good with an LCD display, I'll attach the hex for you to try.

Hey @3d-gussner, thanks for the resorces, I was able to merge without many issues and I already have made a PlatformIO script to build each or all variants configured.

TheZeroBeast, 2020-06-02:
All these changes are already implemented and released.

closed 2020-06-02 by TheZeroBeast